gridhead/nvidia-auto-installer-for-fedora-linux #3: No way to detect host is really Fedora

t0xic0der, 2020-05-22:
For now, it is up to user's discretion as to where they would want to use this tool. There is no way to check the host OS and tell if it is Fedora or not.

ferdnyc, 2020-05-26:
Putting aside the question of whether you really _want to_ detect such things (why shouldn't the tool to be runnable on CentOS, or RHEL, or anywhere else the necessary dependencies are supported?), it's definitely possible. With `python3-distro` installed from the Fedora repos:
```python3
>>> import distro
>>> distro.linux_distribution()
('Fedora', '32', '')
```

t0xic0der, 2020-05-26:
That is pretty clean. I would use that.

Coming to the part of making the tool run on a distribution having all the neccessary dependencies, it is totally possible to do so but it would not be of much use there. I don't think RPMFusion acts as a provider to CentOS and/or RHEL (or do they?).

But if it is executed on distributions in the likes of Ubuntu, Arch Linux etc., it should let know the user from the get-go that by detecting the distribution that the tool is not going to work for them. (But if this becomes an nonfree package - chances of people running it there are pretty low)

ferdnyc, 2020-05-26:
> I don't think RPMFusion acts as a provider to CentOS and/or RHEL (or do they?).

Yup, in both cases: https://rpmfusion.org/Configuration
(CentOS can use the same repos as the equivalent RHEL release)

I'm almost certain they don't ship with `rpmfusion-nonfree-nvidia-driver` repo configs preinstalled, though, so `RepoInstaller.py` would need some changes. Executing some of the setup steps on that Configuration wiki page would be a prereq. The nvidia drivers are in the `rpmfusion-nonfree` repo on EL7/8. (They are on Fedora, as well, just that they don't need to be pulled from there thanks to the bundled driver-only repo.)

> But if it is executed on distributions in the likes of Ubuntu, Arch Linux etc., it should let know the user from the get-go that by detecting the distribution that the tool is not going to work for them. 

Agreed. Best way to do _that_ is probably by checking for the presence of the `dnf` command, and bailing if it's not found. You can't do anything on a system without a working `dnf`. Whereas if it's useful on the current system, there's at least a fair chance the tool is as well.

closed 2020-05-27 by t0xic0der