gridhead/nvidia-auto-installer-for-fedora-linux #13: Facilitate removal of leftovers from previous unused kernel modules

t0xic0der, 2020-05-23:
[image]

ferdnyc, 2020-05-26:
Personally I would strongly caution against this.

If akmods are going to be cleaned up automatically, the `akmods` script/service should be the one to do it. It's not the job of an installer tool to mess around inside the cache directory of some completely _different_ tool. That's not in any way an essential step in the process of installing display drivers, and there's no justification for it in terms of it solving problems or whatever. It's doing housekeeping in someone else's house.

It's fine if someone wants to add that to a personal script they write, but not in a tool that's meant to be distributed to others.

There's already an [open bug report](https://bugzilla.redhat.com/show_bug.cgi?id=1542658) against akmods regarding its lack of cleanup, and if a fix is to be implemented, it should be implemented there.

t0xic0der, 2020-05-26:
:+1: Got it. Let us keep this only to the installation part.

Also, I have come to notice then when Fedora Workstation (and maybe the other spins too) install a new kernel when they have an excess of five kernels already - they remove the oldest one. With the removal of the oldest of the installed kernel, packages dependent of it get removed too. You find an `akmod` kernel module build there for the older kernel in the to-be-removed list.

It is kind of an indirect way but `dnf` seems to clean up stuff, once that kernel is removed but the same is left back for the remaining four kernels that are not currently in use. Let us leave this cleaning up stuff to be done by [them](https://bugzilla.redhat.com/show_bug.cgi?id=1542658) only as you mentioned.

ferdnyc, 2020-05-26:
> Got it. Let us keep this only to the installation part.
>
> Also, I have come to notice then when Fedora Workstation (and maybe the other spins too) install a new kernel when they have an excess of [...]

Well, that's actually kind of the crux of the problem. Yes, the _installed_ packages get removed — but the _built_ packages in `/var/cache/akmods/` **don't**. New builds of every driver release against every one of the installed kernels will just keep accumulating there, literally forever. I tend to clean mine out around twice a year, usually when I'm looking to do a `dnf system-upgrade` and I'm hurting for space to store the upgrade packages. (I made the root partition on my primary machine _way_ too small.)

But, despite that, it's still `akmods`' mess to clean up (or not).