gridhead/nvidia-auto-installer-for-fedora-linux #14: Add support on proprietary drivers for older discrete cards

t0xic0der, 2020-05-23:
[image]

ferdnyc, 2020-05-25:
Quadro K2100M is on the [390.132 support list](https://download.nvidia.com/XFree86/Linux-x86_64/390.132/README/supportedchips.html), it's in the same boat as anyone with a GeForce card older than the GT 630, and comes with all the same problems of having to figure out that the older drivers need to be installed.

Perhaps the simplest thing, as unsatisfying as it is from a user-interface perspective, is just to either supply separate command aliases (`nvidia-autoinstaller-for-fedora-ws-390xx`), or take an argument specifying which legacy driver release to use (`nvidia-autoinstaller-for-fedora-ws --driver=390xx` or `nvidia-autoinstaller-for-fedora-ws --legacy|--legacy340` or something).

Either requires the user to _know_ that they need older drivers, which is unfortunate, but that's no different from the current situation. It'll remain a hard problem to solve, as long as Nvidia doesn't make it even remotely easy to determine which drivers are needed programmatically.

But other than that, it should be possible to use all of the same commands, just adding `-390xx` or `-340xx` to the appropriate package names  in the `dnf` transactions as needed. The RPM Fusion packages are all set up the same regardless of driver release, so once that can be specified and it's incorporated into the package selection, users with legacy cards should be able to make use of the tool.

onuralpszr, 2023-01-14:
@t0xic0der  We can get "old card"  vendor product list as possible as we can and add manual option about "install certain releases based on rpmfusion docs and RPM packages they provide 

https://rpmfusion.org/Howto/NVIDIA?highlight=%28%5CbCategoryHowto%5Cb%29

BUT, we need give some fair warnings about "Supported on current stable Xorg server release. EOL by NVIDIA at the end of 2019. Still available on "best effort basis" (newer kernel may break, will be discontinued at anytime if not actively maintained)" 

Because VERY old cards not gonna work and high chance It can break X or Wayland session so best we can do this "we can add this "legacy" parameters options first then we can try automatize by checking vendor and product ID  OR by checking Architecture code name like GA10X or AD1XX  (list : https://endoflife.date/nvidia-gpu)  and determine old ones and new ones too, we may need to search bit more but it is possible to certain list to make bit more easier to everyone to install automatically.