gridhead/nvidia-auto-installer-for-fedora-linux #16: Port this to RHEL/CentOS if possible as a project

t0xic0der, 2020-05-23:
[image]

lcts, 2020-07-20:
I'd like to help, but don't have the time right now. So if someone wants to take a look at this:

The script already builds without issue for [both EPEL7 & 8](https://copr.fedorainfracloud.org/coprs/lcts/nvautoinstall/build/1565096/), so at least all the script's dependencies are satisfied. So to get this working on RHEL/CentOS, initially one would have to
 - detect if on CentOS/RHEL
 - replace hardcoded dnf calls with variable ones that can handle both dnf & yum
 - check if dnf is installed, use yum if not
 - enable EPEL repositories & any other prereqs for [RPMFusion if on RHEL/CentOS](https://rpmfusion.org/Configuration)
 - replace hardcoded repository URLs with variable ones taking into account the different distributions

The rest can most likely stay unchanged, unless some needed packages are missing/named differently. But since most of the stuff going on here is based on RPMFusion/NVidia repos, that seems unlikely.

t0xic0der, 2020-07-26:
Apologies for the delayed response from my end. College life is getting difficult (and it always it - don't let anyone tell you any different :laughing:)

> detect if on CentOS/RHEL

I am using the `distro` package in Python for detecting the distribution name and all other attributes. The same can be put to use for detecting CentOS/RHEL the very same way they are used for detecting specifically Fedora.

> replace hardcoded dnf calls with variable ones that can handle both dnf & yum

Now that's where we can avoid all the hard work but simply creating yet another binary with support for RHEL. Using CLI composers as `click` can indeed make it simple while keeping the complexity of having both in the same binary off the scene.

> check if dnf is installed, use yum if not

An intelligent thing to do. Currently, I straightaway assume that `dnf` is available and I am always checking for exit codes from the backend to figure if things are going right or not. Should `dnf` is not found, this always gets out with a non-zero exit code which then helps to exit the program prematurely as the error cannot be recovered from during runtime.

> enable EPEL repositories & any other prereqs for RPM Fusion if on RHEL/CentOS

I need to take a look at the RHEL/CentOS repositories now for I do not have much information on them. But thanks for pointing them out. They really need to be considered.

> replace hardcoded repository URLs with variable ones taking into account the different distributions

CUDA is the only repository that is fetched beyond RPM Fusion's view - and the website itself states its compatibility for all RPM-based distributions. Indeed there can be issues where either `dnf` or `yum` needs to be chosen from but the binaries fetched from the repository do stay the same.

> The rest can most likely stay unchanged, unless some needed packages are missing/named differently.

One of the most important issues by the way. Package names can indeed be different and one cannot be sure of it unless one installs drivers the hard way on RHEL/CentOS. (Non-issue for RPM Fusion but a big one if some other repository is used)

I am trying to get contributors and maintainers for this and I hope for you to do the same. I know it might be a little too much to ask for but owing to the fact that this tool has helped a lot of people so far and would certainly continue to - I believe the project should not go unmaintained.

lcts, 2020-07-26:
The yum/dnf discrepancy will become a non-issue at some point, as RHEL/CentOS8 already uses dnf. But in any case, it's not complicated to work aroun: Put the executable name in a variable, and perform the actual install command with `$varname install ...`, then set varname = dnf/yum as appropriate after detecting the OS.

The CUDA repos [is split between the different distros](http://developer.download.nvidia.com/compute/cuda/repos/), though I haven't checked if the actual binaries are different - probably not. But again, if you've already detected what OS you're on, you can simply put that in a variable and change the url to `'http://developer.download.nvidia.com/compute/cuda/repos/' + os + '/x86_64/cuda-' + os + '.repo'`.

> One of the most important issues by the way. Package names can indeed be different and one cannot be sure of it unless one installs drivers the hard way on RHEL/CentOS. (Non-issue for RPM Fusion but a big one if some other repository is used)

The CUDA/RPM Fusion/EPEL packages will be named identically, so one would only have to check if their dependencies in normal os repos are different. I'll try to get on to that.

lcts, 2020-07-27:
OK, so I had a look at the direct dependencies, that seems mostly OK. `vulkan` is provided by `vulkan-loader` on fedora and el8, but the package has the appropriate `Provides:` so a switch is not strictly needed (though probably a good idea). For some reason, `libva-utils` is missing in EPEL8. The other deps are fine.

NicTanghe, 2021-02-03:
Anny ETA on this ?

t0xic0der, 2021-02-03:
Hello,

I am yet to investigate the nuances to be considered and the dependencies
to include when implementing the tool in RHEL8/CentOS8/CentOS Stream etc.
Feel free to join the development to make the port happen sooner.

Thanks,

NicTanghe, 2021-02-03:
I`ll gladly help but i mostly need this becouse i always fail at installing video drivers.
i`m about to start following these instructions for RHEL8 
https://access.redhat.com/solutions/4134381

I think they might help figuring out variables.

t0xic0der, 2021-02-03:
The article that you have provided is not generally accessible without a
Red Hat account. Could you share some other references which are
accessible?

Maybe consider sending a PDF of the webpage that this link opens up.

NicTanghe, 2021-02-03:
I can do you one better.
Just get a dev acount here.
https://developers.redhat.com/articles/getting-red-hat-developer-subscription-what-rhel-users-need-know

t0xic0der, 2021-02-03:
I am currently not interested in making a developer account. Please provide
an alternative documentation for reference.

NicTanghe, 2021-02-03:
Resolution

Note: Depending on the GPU, you may not have graphics at all to start, so you will need to download the NVIDIA driver for your GPU on another computer, and transfer it to the target machine via alternative means. At the time of this writing, this will be the case with Volta/Turing family GPUs.

Note: At the time of writing, the NVIDIA proprietary driver does not work with Wayland. You must install X11 for this to work; steps included below.

Note: On Volta/Turing GPUs (RTX series) you probably set nomodeset on the grub config line at installation time. You will need to revert this setting and rebuild the grub configuration after completing the driver installation, otherwise, the NVIDIA driver will not work.

Remove nomodeset from the kernel cmdline in /etc/default/grub.

    For legacy (BIOS mode) installations:
    Raw

    # grub2-mkconfig -o /boot/grub2/grub.cfg

    For EFI systems:
    Raw

    # grub2-mkconfig -o /boot/efi/EFI/redhat/grub.cfg

Installation Steps

    Acquire the latest NVIDIA driver from https://www.nvidia.com and place it in root's home directory.

    Make the file executable:
    Raw

    # chmod +x NVIDIA-$version.sh

    Blacklist the nouveau module:
    Raw

    # echo 'blacklist nouveau' >> /etc/modprobe.d/blacklist.conf

    Install dependencies (replace "Server with GUI" with "Workstation" for RHEL 8 Workstation):
    Raw

    # dnf groupinstall "Server with GUI" "base-x" "Legacy X Window System Compatibility" "Development Tools"
    # dnf install elfutils-libelf-devel "kernel-devel-uname-r == $(uname -r)"

    Back up and rebuild your initramfs:
    Raw

    # mv /boot/initramfs-$(uname -r).img /boot/initramfs-$(uname -r)-nouveau.img
    # dracut -f

    Change the default runlevel:
    Raw

    # systemctl set-default multi-user.target

    Reboot the system:
    Raw

    # reboot

    Install the driver:
    Raw

    # ./NVIDIA-$version.run

    Test the new driver:
    Raw

    # systemctl isolate graphical.target

    If the test is successful, correct your default runlevel:
    Raw

    # systemctl set-default graphical.target

t0xic0der, 2021-02-03:
Thanks. I'd go through this writeup and see if I can help.

NicTanghe, 2021-02-03:
THX you a hero. installing gfx drivers is always the biggest hurdle 4 me.

t0xic0der, 2021-02-03:
You're welcome. I'd keep you posted about the progress made.

NicTanghe, 2021-02-03:
Oh well following the guide doesn`t work for me becouse the driver doesn`t compile.
Then i found this.
https://developer.nvidia.com/blog/streamlining-nvidia-driver-deployment-on-rhel-8-with-modularity-streams/
I`ll keep u updated.