gridhead/nvidia-auto-installer-for-fedora-linux #5: Generalize and test the tool in KDE Plasma spin

t0xic0der, 2020-05-23:
Need contributors to test and confirm the tool's workability in KDE Plasma spin.

t0xic0der, 2020-05-23:
Get the distro [here](https://spins.fedoraproject.org/en/kde/).

shirishgoyal, 2020-06-11:
I tested the KDE Plasma spin and it worked fine. Only additional change I had to do was to turn SELinux to permissive mode.

While it was in Enforcing mode, there was an error while installing akmod which despite all attempts was not working. It always resulted in below error:
useradd:libbz2.so.1: failed to map segment from shared object

I tried installing libbz2-devel lib but it didn't solve the issue and "useradd" part seemed like permission issue and further exploration pointed towards SELinux.

t0xic0der, 2020-06-11:
Thanks for your attention in this issue. Does the SELinux needs to stay in permissive all the time or is it required only during the installation process?

t0xic0der, 2020-06-11:
Also there are probably two ways of dealing with this. One is where we leave behind your comments in the documentation such that people using KDE spin can adhere to it as a general fix. Other is to actually explore in depth the causative agent of this issue and deduce a specific fix for it.

Would you be wiling to contribute to this project? If so, I will assign you to this issue.

shirishgoyal, 2020-06-12:
I don't think it is a KDE specific issue. In your vanilla Fedora 32, what was the default status of SELinux?

t0xic0der, 2020-06-12:
I did not bother to check the SELinux status on Workstation as the tool worked flawlessly anyway. I wish I had a copy of Fedora 32 Workstation lying around so that I could execute a quick `sestatus` to check it.

karlisk, 2022-04-13:
Have been running Fedora 34 and now 35 the KDE Spin, this tool has been very helpful to wrangle Nvidia PRIME. Ran into a few issues (specifically with PRIME, of course) but for the most part I have all the autoinstaller components installed and everything works without any additional configuration needed. But because of the aforementioned problems (or rather hinderance) I thought I'd add some feedback.

```
❯ cat /etc/*release
Fedora release 35 (Thirty Five)
NAME="Fedora Linux"
VERSION="35 (KDE Plasma)"
ID=fedora
VERSION_ID=35
VERSION_CODENAME=""
PLATFORM_ID="platform:f35"
PRETTY_NAME="Fedora Linux 35 (KDE Plasma)"
ANSI_COLOR="0;38;2;60;110;180"
LOGO=fedora-logo-icon
CPE_NAME="cpe:/o:fedoraproject:fedora:35"
HOME_URL="https://fedoraproject.org/"
DOCUMENTATION_URL="https://docs.fedoraproject.org/en-US/fedora/f35/system-administrators-guide/"
SUPPORT_URL="https://ask.fedoraproject.org/"
BUG_REPORT_URL="https://bugzilla.redhat.com/"
REDHAT_BUGZILLA_PRODUCT="Fedora"
REDHAT_BUGZILLA_PRODUCT_VERSION=35
REDHAT_SUPPORT_PRODUCT="Fedora"
REDHAT_SUPPORT_PRODUCT_VERSION=35
PRIVACY_POLICY_URL="https://fedoraproject.org/wiki/Legal:PrivacyPolicy"
VARIANT="KDE Plasma"
VARIANT_ID=kde
Fedora release 35 (Thirty Five)
Fedora release 35 (Thirty Five)
```

As I mentioned - I installed all components step by step and everything, except the PRIME components installation worked fine without any additional interference. With PRIME I had an issue where built-in display didn't show anything but a black screen, undoing the install fixed the issue  but then with HDMI cable attached it revealed a massive performance drop (graphical interface would become noticably slowed down), detaching the HDMI immediately solved the performance drop. With PRIME components installed, however, and an external display attached via HDMI it revealed that black screen only affects the built-in display, while external display works just fine (no performance issues either). And post log-in the display issues seem to be fixed (indicating it's a SDDM related issue), after a quick Googling around lead me to [this](https://ask.fedoraproject.org/t/optimus-kde-on-nvidia-descrete-card-black-screen/6924
/etc/sddm/Xsetup).

The solution in my case required 'xrandr' to be installed because, on a clean Fedora 35 install, it was not installed otherwise. Then what fixed the issue with black screen during login screen was appending (as indicated in the Arch wiki and the askfedora post):
```
xrandr --setprovideroutputsource modesetting NVIDIA-0
xrandr --aut
```
to /etc/sddm/Xsetup   ... with this I have all the components of the autoinstaller installed and in working order, GWE (GreenWithEnvy) util also seems to be working just fine

QazCetelic, 2022-07-07:
#110 The SDDM does not display.

OriCat101, 2023-06-19:
The driver installation worked fine, however If install primec it tells me it was successful and to reboot. However the prime still does not appear to be working, I checked in  "NVVIDIA X Server settings"

OS: Fedora Linux 38 (KDE Plasma) x86_64
Host: ROG Flow X13 GV301RE_GV301RE 1.0
Kernel: 6.3.8-200.fc38.x86_64
CPU: AMD Ryzen 7 6800HS with Radeon Graphics 
GPU: NVIDIA GeForce RTX 3050 Ti Mobile
GPU: AMD ATI Radeon 680M
Memory: 3402MiB / 15212MiB