gridhead/nvidia-auto-installer-for-fedora-linux #38: Add automatic detection and configuration for Legacy Geforce 8/9/200/300

t0xic0der, 2020-06-01:
Specific support is needed for
- Proprietary 340.xx drivers for the legacy discrete cards
- Proprietary CUDA support upto 6.5 for the legacy discrete cards

viniciusferrao, 2021-06-26:
Hi @t0xic0der there's any way to force it right now? I'm trying to use the software to install the drivers of a GeForce 9400M in a 2009 Mac Mini.

t0xic0der, 2021-06-27:
@viniciusferrao,

This would require reading the output of the following commands first,

```
$ lspci | grep "VGA"
```

and 

```
$ lspci | grep "3D"
```

The example output for the first one is

```
00:02.0 VGA compatible controller: Intel Corporation HD Graphics 630 (rev 04)
```

and the same for the second one is

```
01:00.0 3D controller: NVIDIA Corporation GP107M [GeForce GTX 1050 Mobile] (rev a1)
```

Then applying a custom regular expression filter to differentiate between the cards and their variants. 

As I am not able to devote enough time to write regex, for now, I would most likely be able to come to fix this some months later.

Alternatively, please feel free to open a pull request for the same and I would be glad to include it.

@shivangswain, now that you're practicing regular expression nowadays - could you please help us out on this one?