gridhead/nvidia-auto-installer-for-fedora-linux #41: v0.25 RPM fusion NVIDIA chould not be enable

Crimsonfantasy, 2020-06-01:
hi, 
i have a issue when installing. Here is my prompt message:
`[ # ] NVIDIA AUTOINSTALLER FOR FEDORA 32 AND ABOVE
[ ✘ ] Installer cannot proceed without root privileges
[ ✘ ] Leaving installer
[fenrir@localhost Downloads]$ sudo ./NVAutoInstFedora32 
[sudo] password for fenrir: 
[ # ] NVIDIA AUTOINSTALLER FOR FEDORA 32 AND ABOVE
[ # ] GATHERING ACTIVE INFORMATION...
[ ✔ ] Host information was gathered
      System: Linux v5.6.14-300.fc32.x86_64
      Hostname: localhost.localdomain
      Version: #1 SMP Wed May 20 20:47:32 UTC 2020
      Distribution: Fedora 32 x86_64
[ ✔ ] Supported OS detected - Fedora 32 x86_64
[ # ] CHECKING FOR GPU COMPATIBILITY...
[ ✔ ] Compatibility infomation was obtained
      00:02.0 VGA compatible controller: Intel Corporation HD Graphics 630 (rev 04)
      01:00.0 VGA compatible controller: NVIDIA Corporation GP107M [GeForce GTX 1050 Mobile] (rev a1)
[ ✔ ] An active NVIDIA GPU was detected!
[ ✔ ] An Optimus Dual GPU setup was detected!
[ # ] LOOKING FOR EXISTING DRIVER PACKAGES...
[ ! ] No existing NVIDIA drivers were detected!
[ ! ] Do you wish to install the drivers? (Y/N) y
[ # ] ATTEMPTING CONNECTION TO RPM FUSION...
[ ✔ ] Connection to RPM Fusion server was estabilished
[ ! ] RPM Fusion repository for Proprietary NVIDIA Driver not detected
[ ! ] Do you wish to fetch packages from this repository? (Y/N) y
[ # ] FETCHING REPOSITORY DATA...
[ ✘ ] RPM Fusion NVIDIA repository could not be enabled
[ ✘ ] Leaving installer

How do I fixed it?
Thank you.`

Crimsonfantasy, 2020-06-01:
hi i am the questioner, 
I found the solution by manual enable Nvidia fusion:
The guideline is written in here:
[https://ask.fedoraproject.org/t/optimus-setting-the-nvidia-gpu-as-primary-rpmfusion-in-fedora-32-workstation/6550/3](how to enable Nvidia  fusion)

t0xic0der, 2020-06-01:
Which version of Fedora are you using? Is it Workstation or one of the spins?

t0xic0der, 2020-06-01:
Don't bother answering that. It indeed was a bug. I tested it on Fedora Workstation and KDE spin and found it. I have fixed it in the [latest merge](https://github.com/t0xic0der/nvidia-auto-installer-for-fedora/commit/50f5654a97ba986a6b0cfd2a6036f6c8345f2aed) and will be making a release with the bug fix shortly.

Thank you for marking out the issue. @Crimsonfantasy

t0xic0der, 2020-06-01:
Check this [version](https://github.com/t0xic0der/nvidia-auto-installer-for-fedora/releases/tag/v0.2.6). The issue should be resolved now.

closed 2020-06-01 by t0xic0der