gridhead/nvidia-auto-installer-for-fedora-linux #44: dnf update failed : conflict vidia-kmod-common-3:440.95.01-1.fc29.x86_64

sylvoie, 2020-06-28:
Hello,

First, thank you very much for this super tool.

This is what I did :

- Fresh installation of fedora 32.
- SELinux off (same problem than issue : "Generalize and test the tool in KDE Plasma spin #5")
- dnf update
- installed nvidia driver (all steps)
- reboot
- installed some tools like VSC.
- reboot
- installed KDE and Xfce
- reboot
- dnf update (error)

I received that message :

Erreur : Erreur de la transaction de test :
  le fichier /usr/lib/udev/rules.d/60-nvidia.rules de l'installation de nvidia-kmod-common-3:**440.95.01-1.fc29**.x86_64 entre en conflit avec le fichier du paquet xorg-x11-drv-nvidia-3:**440.82-1.fc32.x86_64**

I don't understand that my update try to update from fc32 to nvidia driver fc29?

I'm stuck and unable to update my environment.

Thank's to help me

Sylvain

t0xic0der, 2020-06-28:
> Fresh installation of fedora 32.

Fedora Workstation or any one of the spins? (Or as told by you later, specifically KDE Plasma spin?) The question might sound irrelevant but it still can give leads to me as to where I can start reproducing this bug.

> SELinux off (same problem than issue : "Generalize and test the tool in KDE Plasma spin #5")

The tool worked on devices I tested (both laptop and desktop PCs having Workstation 32) without actually needing to tweak the SELinux settings. It might be necessary for the KDE Plasma spin as mentioned in #5.

Plus, it does not matter if you fetched the updates first or not as the tool would do that for you, if you haven't. Also, it would check for active internet connection too and there is no need to have the drivers installed first and then other applications and stuff (no chronology stuff needed here - the tool simply fetches proprietary drivers from the RPM Fusion repositories and installs them), so you can do it anytime.

> Erreur : Erreur de la transaction de test :
le fichier /usr/lib/udev/rules.d/60-nvidia.rules de l'installation de nvidia-kmod-common-3:440.95.01-1.fc29.x86_64 entre en conflit avec le fichier du paquet xorg-x11-drv-nvidia-3:440.82-1.fc32.x86_64

This thing really dazzles me. If you have a fresh installation of Fedora 32, there is literally no reason why it would fall back into even checking with the Fedora 29 repos. The tool was explicitly designed for Fedora 32 Workstation and is reported to be working in the spins as well. You might want to give a fresh installation of the KDE Plasma spin again (if that is what you'd use) and use steps listed in #5 during the installation.

If you make any other observation as to why this might have happened, please let know here. It will be some time before we get people to actually contribute by testing it on spins. There are open issues for that. :slightly_smiling_face:

sylvoie, 2020-06-28:
Hello,

> Fresh installation of fedora 32.

Fedora workstation 32 base on Gnome that I downloaded last friday. Yes I will probably try the "KDE Plasma spin" later.

> SELinux off (same problem than issue : "Generalize and test the tool in KDE Plasma spin #5")

I understand, and I tried several times with always the same result when SELinux is enabled. SELinux was "enforcing" and I need to disable to successfully install.

> Erreur : Erreur de la transaction de test :
> le fichier /usr/lib/udev/rules.d/60-nvidia.rules de l'installation de nvidia-kmod-common-3:440.95.01-1.fc29.x86_64 entre en conflit avec le fichier du paquet xorg-x11-drv-nvidia-3:440.82-1.fc32.x86_64> 

Before using "KDE Plasma spin", I will reinstall "Fedora workstation 32" without installing KDE and XFCE to see.

Thank's

Sylvain

sylvoie, 2020-06-28:
Hello,

I reinstalled from scratch and followed these steps :

- Install Fedora 32 workstation
- sudo yum update
- SELinux = permissive
- reboot
- sudo ./NVAutoInstFedora32 --rpmadd
- sudo ./NVAutoInstFedora32 --driver
- sudo ./NVAutoInstFedora32 --x86lib
- sudo ./NVAutoInstFedora32 --nvrepo
- sudo ./NVAutoInstFedora32 --ffmpeg
- sudo ./NVAutoInstFedora32 --vidacc
- reboot
- sudo ./NVAutoInstFedora32 --plcuda
- sudo ./NVAutoInstFedora32 --vulkan
- reboot
- sudo yum update

Last yum update result :

==================================================================
 Paquet                                       Architecture                    Version                                       Dépôt                         Taille
==================================================================
Mise à jour:
 akmod-nvidia                                 x86_64                          3:440.95.01-1.**fc29**                            **cuda**                           13 M
Installation des dépendances:
 nvidia-kmod-common                           x86_64                          3:440.95.01-1.**fc29**                            **cuda**                           10 k
Ignorer les paquets en conflit :
(ajouter « --best --allowerasing » à la ligne de commande pour forcer leur mise à niveau):
 nvidia-driver-cuda                           x86_64                          3:440.95.01-1.**fc29**                            **cuda**                          313 k
Ignorer les paquets ayant des dépendances cassées :
 nvidia-driver                                x86_64                          3:440.95.01-1.**fc29**                            **cuda**                          2.3 M
 nvidia-persistenced                          x86_64                          3:440.95.01-1.**fc29**                            **cuda**                           91 k

Résumé de la transaction
==================================================================
Installer        1 Paquet
Mettre à niveau  1 Paquet
Ignorer          3 Paquets

Did I do something wrong?

Why these fc29 come from cuda repository?

Is-it something with the french installed language?

Thank's

Sylvain

t0xic0der, 2020-06-28:
> Why these fc29 come from cuda repository?

That is okay. CUDA is not actively maintained but the thing made in F29 works just as great in F32.

> Is it something with the french installed language?

Definitely not. French is one of the leading languages in Weblate with much of the part already translated and accommodated to.

CUDA might just be the core reason behind this. As the F29 CUDA repos require drivers exclusively for F29, it is turning up even for `akmods-nvidia` package. There is nothing wrong with the package though but I believe F29 is not maintained anymore so using their repositories is not advised.

t0xic0der, 2020-06-28:
The CUDA problem is definitely an issue with the RPM package available in the official RPM Fusion repositories and not with the installer (as it simply automates the steps listed [here](https://rpmfusion.org/Howto/NVIDIA) and [here](https://rpmfusion.org/Howto/CUDA) in a convenient way. The CUDA package is not up-to-date and pulls outdated packages as its dependencies.

sylvoie, 2020-06-30:
Hello,

I agree with you, that problem is not with the installer. Thank you for this installer, it saved my personal project to run first in Linux.

I installed and I confirm that this installer works well with Fedora 32 spins Xfce without CUDA (--nvrepo --plcuda). I can install CUDA, and that is working fine, but after that updates are stuck because CUDA package is not up-to-date.

At this time, I have driver 440.82, dnf tries to install 440.95 with CUDA Repo. On NVidia site, the driver are at 440.100. snif!

With my research, I found this that I share with you : https://www.youtube.com/watch?v=IVpOyKCNZYw.

Thank's

Sylvain

t0xic0der, 2020-09-23:
> At this time, I have driver 440.82, dnf tries to install 440.95 with CUDA Repo. On NVidia site, the driver are at 440.100. snif!

The differences between the versions made available on the NVIDIA site and on the RPM Fusion repositories is due to the fact that recompiling the versions acquired from the NVIDIA site into an RPM package takes time, which then creates an offset. This should not be a concern though as you would always have a slightly older but rock stable drivers as a result of Fedora-specific configurations and thorough testing made before release by the folks at RPM Fusion.

> With my research, I found this that I share with you : https://www.youtube.com/watch?v=IVpOyKCNZYw.

Thanks for the research. :rofl:  

In a matter of the next few days, a new version of the tool would roll out packaged in a COPR which would have a newer CUDA repository (the one which would not hold you back into using dependencies from as old as F29). I would be modifying the instructions and releases shortly so stay tuned.

closed 2023-01-13 by onuralpszr