gridhead/nvidia-auto-installer-for-fedora-linux #52: Add CONTRIBUTING.md and CODE_OF_CONDUCT.md

t0xic0der, 2020-09-25:
In the spirit of getting more contributors for the project, the above documents would be appreciated.

codePerfectPlus, 2020-09-25:
@t0xic0der I like to be assigned for the adding `CONTRIBUTING.md` docs.

t0xic0der, 2020-09-25:
Thank you for considering the issue, @codePerfectPlus. If you are considering the issue, please consider it wholly and add CODE_OF_CONDUCT.md as well.

codePerfectPlus, 2020-09-25:
Yeah I will add CODE_OF_CONDUCT.md too.

t0xic0der, 2020-09-25:
Cool. Thanks. :smile:

t0xic0der, 2020-09-29:
Hi @codePerfectPlus, Be sure to get yourself registered at https://organize.mlh.io/participants/events/4659-hack-astrosonic, our official Hacktoberfest event to stay in touch with the fellow contributors of this project.

Also, feel free to leave a star on the project - should you find it worth your time. Happy hacking! :smile:

t0xic0der, 2020-10-15:
Closing this issue as it was addressed with #53.

closed 2020-10-15 by t0xic0der