gridhead/nvidia-auto-installer-for-fedora-linux #54: symbolic link vdpau

boistordu, 2020-09-29:
Not sure it concerns your script because I don't recall if your script does take into account the vdpau plugin. 
But What I can see on my install after a few upgrades through dnf, is that the symbolic link of /usr/bin/libvdpau_nvidia.so doesn't point to the correct lib 
here : lrwxrwxrwx.  1 root root       31 Jul 21 21:01 libvdpau_nvidia.so -> vdpau/libvdpau_nvidia.so.440.36

and should be: 
lrwxrwxrwx.  1 root root       31 Jul 21 21:01 libvdpau_nvidia.so -> vdpau/llibvdpau_nvidia.so.450.66 
or 
the generic symbolic link already in place : 
lrwxrwxrwx.  1 root root       31 Jul 21 21:01 libvdpau_nvidia.so -> vdpau/llibvdpau_nvidia.so.1

t0xic0der, 2020-09-29:
You guessed it right. My script does not really consider or make such changes. You would want to report the same to Fedora's bugzilla under the akmods package I think.

boistordu, 2020-09-30:
I need to put a reminder on my to-do list because I m not sure I already did report to them anything so I will first need to check which website is it etc and I have a lot of things to do first. Could you ping me sometime to do it? Unless you have already all the right links and so that it can speed up my process

t0xic0der, 2020-09-30:
Here you go. https://bugzilla.redhat.com/.

I am completely packed up this October for I am working on representing Fedora for this Hacktoberfest. Do register for the event. https://organize.mlh.io/participants/events/4779-foss-and-fedora. :smile: