gridhead/nvidia-auto-installer-for-fedora-linux #55: Fedora 33 support

erdius, 2020-10-03:
The tool is currently not working of Fedora 33 beta

t0xic0der, 2020-10-03:
Thanks for marking this out. Could you please tell what problems you are facing?

ediX0, 2020-10-04:
Same issue here,  i have tested the tool before on Fedora 32 so i know it works on my hardware.
No errors while running it after the reboot i'm unable to launch Nvidia X Settings.

[ ✔ ] Driver package installation completed!
[ ! ] Commencing mandatory sleep for 5 minutes to load up kernel modules
[ ✔ ] Modified kernel modules have been loaded up!
[ ✔ ] FORCING READ FROM NEW KERNEL MODULES...
Checking kmods exist for 5.8.13-300.fc33.x86_64            [  OK  ]
      Read modified kernel module progressing [1/2]
      Read modified kernel module progressing [2/2]
[ ✔ ] Kernel module installation completed!
[ ! ] Commencing mandatory sleep for 3 minutes to save kernel read config
[ ✔ ] Kernel read configs have been stored!
[ ✔ ] DRIVER INSTALLATION SUCCESSFULLY COMPLETED!

Running nvidia-settings from term gives: ERROR: Unable to find display on any available system.

Anything I can do to help you troubleshoot ?

t0xic0der, 2020-10-04:
@ediX0 Please use the most recent version of the tool. That does away with the mandatory sleep times. Also, is it failing on F32 or F33 Beta for you?

erdius, 2020-10-04:
I think there was a problem with Nvidia drivers a few days ago, i reran the tool today and at least it didn't error out and seemed to install correctly, but I still don't see the driver being loaded on an optimus laptop after a reboot. 

`$glxinfo | egrep "OpenGL vendor|OpenGL renderer"
OpenGL vendor string: Intel
OpenGL renderer string: Mesa Intel(R) UHD Graphics 620 (WHL GT2)
`

t0xic0der, 2020-10-05:
For PRIME support, I wrote a guide [here](https://docs.fedoraproject.org/en-US/quick-docs/how-to-set-nvidia-as-primary-gpu-on-optimus-based-laptops/). Check if it helps you.

P.S. The tool simply automates the installation of NVIDIA drivers but Optimus configurations differ from device to device so certain attention to know they should be set up is necessary.

ediX0, 2020-10-05:
My issue is also with an optimus laptop, exactly the same problem as erdius.
Guess I will stick with F32 for now as it works automatically by just installing akmod-nvidia.

Can't run "always on Nvidia" because i need the battery life aswell on the go.

ediX0, 2020-10-06:
FYI, tried the guide to force the Nvidia CPU by the guide and this doesnt seem to work either.

t0xic0der, 2020-11-11:
> FYI, tried the guide to force the Nvidia CPU by the guide and this doesnt seem to work either.

Hrm strange. Could you detail on what exactly went wrong?

boistordu, 2021-04-29:
Hi,

I'm going to test it with fedora 34 today, but before that I've seen that there was an update for cuda for fedora 33 so I will upgrade to that before to make the upgrade then we'll see what happens.

t0xic0der, 2021-04-29:
Thanks @boistordu. Do let me know of the results. :)

boistordu, 2021-04-29:
still on fedora 33
don't really understand why `sudo dnf -y module install nvidia-driver:latest-dkms`
and why a simple dnf update --refresh would install the new version of cuda if the previous would not be needed. I guess this is because of the installation of xorg which is not what's the command from nvidia is about? 
I didn't reboot. I will do it in a moment .

boistordu, 2021-04-29:
which mode are you recommending by the way to install drivers and cuda and vdaapi? Because it has changed a lot from when I did this 1 year or more agao

boistordu, 2021-04-29:
okey the installation under fedora 34 is completely broken and not functional.
the rpmfusion are activated: 
```
[ # ] NVIDIA AUTOINSTALLER FOR FEDORA
[ ★ ] CHECKING SUPERUSER PERMISSIONS...
[ ✓ ] Superuser privilege acquired
[ ★ ] CHECKING AVAILABILITY OF RPM FUSION NVIDIA REPOSITORY...
[ ! ] RPM Fusion repository for Proprietary NVIDIA Driver was detected
[ ★ ] ATTEMPTING CONNECTION TO RPM FUSION SERVERS...
[ ✓ ] Connection to RPM Fusion servers was established
[ ★ ] LOOKING FOR EXISTING DRIVER PACKAGES...
[ ! ] No existing NVIDIA driver packages were detected
[ ★ ] INSTALLING PROPRIETARY DRIVERS...
Fedora 34 openh264 (From Cisco) - x86_64                                                     434  B/s | 271  B     00:00    
Errors during downloading metadata for repository 'fedora-cisco-openh264':
  - Status code: 404 for https://codecs.fedoraproject.org/openh264/34/x86_64/repodata/repomd.xml (IP: 38.145.60.20)
Error: Failed to download metadata for repo 'fedora-cisco-openh264': Cannot download repomd.xml: Cannot download repodata/repomd.xml: All mirrors were tried
Ignoring repositories: fedora-cisco-openh264
Last metadata expiration check: 0:05:15 ago on Fri 30 Apr 2021 12:50:01 AM CEST.
Package gcc-11.0.1-0.3.fc34.x86_64 is already installed.
Package kernel-headers-5.11.16-300.fc34.x86_64 is already installed.
Package kernel-devel-5.11.15-200.fc33.x86_64 is already installed.
Package kernel-devel-5.11.16-200.fc33.x86_64 is already installed.
Package kernel-devel-5.11.16-300.fc34.x86_64 is already installed.
All matches were filtered out by modular filtering for argument: xorg-x11-drv-nvidia
Error: Unable to find a match: xorg-x11-drv-nvidia
[ ✗ ] Proprietary drivers could not be installed
[ ✗ ] Leaving installer
```
I had to remove every single nvidia packages and now I guess I can't even reboot because I won't have nay graphical UI after that.

boistordu, 2021-04-29:
okey so because of this modular filtering that I do'nt know how to get rid off and which your script would be nice to be able to get rid off, I could not download anythign through your installer and dnf, I had to download everything manually through https://fedora.pkgs.org/34/rpmfusion-nonfree-x86_64/ and wget for each and everyone packages. IT's really upsetting. Did your script activate this filtering on previous version? 

Plus I had to install and improvise with your archive since you didn't provide any warning about the fact that we had now needed to build and install the script in the system....

This script is totally not ready for upgrades. Maybe on fresh installation but clearly not upgrades. 
I had to manually download every single packages listed there: https://ask.fedoraproject.org/t/rpm-nvidia-driver-on-fedora-34-seems-to-have-issues/13742/6

and this is the resulf of a very painful upgrade :+1: https://ask.fedoraproject.org/t/problems-upgrading-33-to-34-workstation/13754/8

t0xic0der, 2021-04-30:
> still on fedora 33
don't really understand why `sudo dnf -y module install nvidia-driver:latest-dkms`
and why a simple `dnf update --refresh` would install the new version of cuda if the previous would not be needed. I guess this is because of the installation of xorg which is not what's the command from nvidia is about?
I didn't reboot. I will do it in a moment .

The dependencies are broken either at RPM Fusion's end or at NVIDIA website's end - and that is why the dependencies for CUDA are not getting refreshed. I will take a look at those in a moment.

> which mode are you recommending by the way to install drivers and cuda and vdaapi? Because it has changed a lot from when I did this 1 year or more agao

For installing these, I would suggest first installing the package from COPR and then running the following commands, one after the other.

```
sudo nvautoinstall --rpmadd
sudo nvautoinstall --driver
sudo nvautoinstall --nvrepo
sudo nvautoinstall --plcuda
sudo nvautoinstall --vidacc
```

> okey the installation under fedora 34 is completely broken and not functional.
the rpmfusion are activated:
> ```
> [ # ] NVIDIA AUTOINSTALLER FOR FEDORA
> [ ★ ] CHECKING SUPERUSER PERMISSIONS...
> [ ✓ ] Superuser privilege acquired
> [ ★ ] CHECKING AVAILABILITY OF RPM FUSION NVIDIA REPOSITORY...
> [ ! ] RPM Fusion [...]

Sorry to hear about that. Seems like the following part of the code failed to execute as the repositories did not have an entry on `xorg-x11-drv-nvidia`, which is very strange as I see that you do have the RPM Fusion repositories activated. Could you please try installing again on a fresh installation?

```python
def main(self):
    exec_status_code = os.system("dnf install -y gcc kernel-headers kernel-devel akmod-nvidia xorg-x11-drv-nvidia xorg-x11-drv-nvidia-libs")
    return exec_status_code == 0
```
> okey so because of this modular filtering that I do'nt know how to get rid off and which your script would be nice to be able to get rid off, I could not download anythign through your installer and [...]

I don't really think my script does anything related to modular filtering. 

> This script is totally not ready for upgrades. Maybe on fresh installation but clearly not upgrades.
I had to manually download every single packages listed there: https://ask.fedoraproject.org/t/rpm-nvidia-driver-on-fedora-34-seems-to-have-issues/13742/6
> 
> and this is the resulf of a very painful upgrade +1 https://ask.fedoraproject.org/t/problems-upgrading-33-to-34-workstation/13754/8

I will check those posts shortly. The most recent changes have been introduced to phase out the executable binaries in favour of pre-packaged releases on COPR. That should be more convenient to users than downloading a binary and executing it. The TARFILE posted in the releases section is for COPR's reference at https://copr.fedorainfracloud.org/coprs/t0xic0der/nvidia-auto-installer-for-fedora/builds/ and not for consumption.

boistordu, 2021-04-30:
Will answer all your request soon but I had a difficult couple of days. 

About the modular filtering I don't know how it got there besides your script, this is my only lead because it's your script that reinstalled properly in fedora 33 in augustus or later maybe my drivers, I think we even have discussed that at the time on the forum. 
The modular filtering have been mentioned several times when I tried to install through dnf the packages.
And as I said it, I had to manually take the download link for each packages in pkg.com and installed it myself. And then just to be sure I've rerun each time your script to see if I didn't forget anything. 

If it is not your script, then it would be great to actually add a coding part to this script about modualr filtering and see there are any activated, and deactivated them completely.

t0xic0der, 2021-05-01:
> If it is not your script, then it would be great to actually add a coding part to this script about modualr filtering and see there are any activated, and deactivated them completely.

Yep. It is something that even @kwizart mentioned in one of his most recent responses.

boistordu, 2021-05-01:
I still didn't have the time to track and understand what's this modular filtering in fedora. It's the first time that I've encountered it in any distributions so I'm a bit frustrated.
But maybe aside of that, or maybe it is linked, I would like more explanations about how the whole system of drivers isntallation + cuda is supposed to go. And then maybe I can put together a more hollistic approach for your README for your repo.
So by starting to my example: 
Now I've followed the one of the post I've linked to you and instlaled one by one the packages. Those packages according to their name are xorg related. I know what's wayland and what's xorg and that they are 2 very different approaches. But cuda, if I let the repo activated is always requesting for deinstalling those packages and replace them with nvidia-drivers and dkms. DKMS I know it's to rebuild at each occurence of a new kernel and add the nvidia drivers to it. 
But what are we supposed to do actually? Since cuda repo is asking different things which reinstall different fc33 packages and have a dkms approach, and here the packages that are working are not close to dkms at all...?

kwizart, 2021-05-01:
Disabling  module is "required" by the official CUDA howto once the nvidia cuda repository is enabled.
https://rpmfusion.org/Howto/CUDA
dnf module disable nvidia-driver
 etc.

Also the openh264 repository failure

boistordu, 2021-05-01:
> Also the openh264 repository failure is on fedora side, its path has to be adapted for f34
@kwizart yeah don't worry about thatI was totally aware of that for a long time now, it just added to the log file that's all.

But then how are we supposed to proceed when installing cuda? Does that mean that we can't have the cuda installed at the same time that the graphic optimization from the nvidia drivers? Or are we supposed to stay with older drivers like fc33 packages and cuda installed even if we upgrade further?

boistordu, 2021-05-07:
okey so according to :
https://dnf.readthedocs.io/en/latest/modularity.html#package-filtering

if I understand correctly, all packages needs to be declared as being part of the module right? 
Well are we sure that the installer is not installing packages that are not compatible with the module in question that could trigger the modularity filtering? 
Do you want another github issue about this?

t0xic0der, 2021-05-09:
> Disabling module is "required" by the official CUDA howto once the nvidia cuda repository is enabled.
https://rpmfusion.org/Howto/CUDA
dnf module disable nvidia-driver
etc.

@kwizart noted. I would promptly introduce the said changes in a quick update. Thanks for marking it out.

@boistordu I wrote a writeup which you can find [here](https://ask.fedoraproject.org/t/solution-modular-filtering-issue-for-nvidia-drivers-cuda-sources-conflict/14063) which has inferences for this issue.

t0xic0der, 2021-08-11:
Closing this issue via #60

closed 2021-08-11 by t0xic0der