gridhead/nvidia-auto-installer-for-fedora-linux #63: Nvidia driver not loading

DSeeLP, 2021-05-26:
I ran
```bash
sudo nvautoinstall --rpmadd
sudo nvautoinstall --nvrepo
sudo nvautoinstall --driver
```
And it said that the driver installed successfully. I restarted my system and found out that the nvidia drivers aren't loaded, instead the nouveau driver was loaded.

How can i fix this?
Im running Fedora 34.
My graphics card is a GTX970.

DSeeLP, 2021-05-26:
Im using a fresh install of Fedora 34 with kernel 5.12.6

HNGamingUK, 2021-10-06:
> Im using a fresh install of Fedora 34 with kernel 5.12.6

Did you ever work out what caused this? I am having the same issue and running 2060 Super.

t0xic0der, 2021-10-06:
Could you please check if you have secure boot enabled?

t0xic0der, 2021-10-06:
Also, I would suggest posting your questions at https://ask.fedoraproject.org/ as the issue is more related to the driver and not with the installer.