gridhead/nvidia-auto-installer-for-fedora-linux #64: Error after installation - Nvidia kernel module missing

arifcelik56, 2021-07-01:
It installed without any problems, but I got this error on the boot. Nvidia app won't open. Im using Lenovo ideapad Gaming 3-15ARH05 Laptop - Amd Ryzen 7 4800H and Gtx1650ti.

Exordio, 2021-07-17:
same error

t0xic0der, 2021-07-18:
Hey @arifcelik56 @Exordio,

Sincere apologies for the delayed reply. Could you please check if the secure boot is turned off or not?

Exordio, 2021-07-18:
Yes, I have already solved my problem. After rebuilding the computer in the BIOS, for some reason, secure boot turned on. 
The funny thing is, I disabled it before and the driver worked fine. In general, what the hell did he turn on, I don't know xd.

arifcelik56, 2021-07-18:
Hi @t0xic0der .
Secure boot was off but I hadn't checked at the time. I deleted it to be able to use the computer but I will install it again soon. I'll post the result here.

t0xic0der, 2021-08-12:
Closing this issue as fixed. For folks who come across a similar issue, try toggling the secure boot to see if it works.