gridhead/nvidia-auto-installer-for-fedora-linux #77: Fedora 35 KDE Plasma: RPM Fusion repository for Proprietary NVIDIA Driver was not found

CleanHit, 2021-11-06:
I'm using Fedora 35 KDE Plasma on a Asus Zephyrus G14 laptop with GTX 1600 TI (Q-Max). I haven't been using Nvidia drives on Linux so far and I'm trying this out for the first time using this repo. When I run `sudo nvautoinstall --rpmadd` I get:
```cmd
[ # ] NVIDIA AUTOINSTALLER FOR FEDORA
[ ★ ] CHECKING SUPERUSER PERMISSIONS...
[ ✓ ] Superuser privilege acquired
[ ★ ] CHECKING AVAILABILITY OF RPM FUSION NVIDIA REPOSITORY...
[ ✗ ] RPM Fusion repository for Proprietary NVIDIA Driver was not detected
[ ✗ ] Leaving installer
```
I do have `rpmfusion-nonfree` and `rpmfusion-free` packages installed

t0xic0der, 2021-11-07:
Thanks for reporting. I'll investigate the issue and get back to you on this.

closed 2021-12-16 by t0xic0der

CleanHit, 2021-12-25:
@t0xic0der thank you for fixing this issue and happy new year!

PR #81 (2021-12-16): Restructuring with Poetry + Modularity + Bug fixes
Fixes #76

Fixes #77

Fixes #79

Fixes #80