gridhead/nvidia-auto-installer-for-fedora-linux #78: Fedora 35 NVIDIA kernel module missing. Falling back to nouveau

YousefSaber, 2021-11-07:
```
(base) [josephsaber@fedora ~]$ screenfetch 
           /:-------------:\          josephsaber@fedora
        :-------------------::        OS: Fedora 
      :-----------/shhOHbmp---:\      Kernel: x86_64 Linux 5.14.16-301.fc35.x86_64
    /-----------omMMMNNNMMD  ---:     Uptime: 1m
   :-----------sMMMMNMNMP.    ---:    Packages: 1855
  :-----------:MMMdP-------    ---\   Shell: bash 5.1.8
 ,------------:MMMd--------    ---:   Resolution: 1920x1080
 :------------:MMMd-------    .---:   DE: GNOME 41.1
 :----    oNMMMMMMMMMNho     .----:   WM: Mutter
 :--     .+shhhMMMmhhy++   .------/   WM Theme: Adwaita
 :-    -------:MMMd--------------:    GTK Theme: Adwaita [GTK2/3]
 :-   --------/MMMd-------------;     Icon Theme: Adwaita
 :-    ------/hMMMy------------:      Font: Cantarell 11
 :-- :dMNdhhdNMMNo------------;       Disk: 442G / 1.4T (32%)
 :---:sdNMMMMNds:------------:        CPU: Intel Core i5-7200U @ 2x 3.1GHz [74.0°C]
 :------:://:-------------::          GPU: Mesa Intel(R) HD Graphics 620 (KBL GT2)
 :---------------------://            RAM: 2355MiB / 11841MiB
                                     
(base) [josephsaber@fedora ~]$ 

```

I disabled secure boot and also followed these [steps](https://ask.fedoraproject.org/t/fedora-34-nvidia-kernel-module-missing-falling-back-to-nouveau/13785/5?u=josephsaber40)
```
(base) [josephsaber@fedora ~]$ rpm -qa akmods
akmods-0.5.6-28.fc35.noarch
(base) [josephsaber@fedora ~]$ --force && sudo dracut --force 
bash: --force: command not found...
(base) [josephsaber@fedora ~]$ sudo akmods --force && sudo dracut --force
[sudo] password for josephsaber: 
Checking kmods exist for 5.14.16-301.fc35.x86_64           [  OK  ]
(base) [josephsaber@fedora ~]$ sudo grubby --update-kernel=ALL --args='nvidia-drm.modeset=1'

(base) [josephsaber@fedora ~]$ cat /etc/gdm/custom.conf
# GDM configuration storage

[daemon]
# Uncomment the line below to force the login screen to use Xorg
WaylandEnable=false

[security]

[xdmcp]

[chooser]

[debug]
# Uncomment the line below to turn on debugging
#Enable=true

```

I used the command 
```
sudo nvautoinstall --driver
```
to install my drivers do you think I should try another option ?

t0xic0der, 2021-11-08:
The driver seems to have installed correctly and that is unfortunately the scope of the project. Could you please open up a topic at https://ask.fedoraproject.org/ on this - there are a lot of folks who can help you with this?

Sincere apologies for not being of much assistance.

YousefSaber, 2021-11-08:
I've already [reported](https://ask.fedoraproject.org/t/nvidia-failure-after-f35-upgrade-geforce-gt740-now-legacy-for-f35/17691) this problem thanks you very much

closed 2021-11-08 by YousefSaber