gridhead/nvidia-auto-installer-for-fedora-linux #82: xorg-x11-drv-nvidia-3:495.46-1.fc35.x86_64 is filtered out

aanno, 2022-01-08:
I'm on fedora 35 and as I'm facing trouble using nvidia with kernel 5.15.11 and 5.15.12 (it is working with 5.15.10). Hence I decided to give nvidia-auto-installer-for-fedora-linux a try.

```bash
$ sudo nvautoinstall driver 
[ # ] NVIDIA AUTOINSTALLER FOR FEDORA LINUX
[ ★ ] CHECKING SUPERUSER PERMISSIONS...
[ ✓ ] Superuser privilege acquired
[ ★ ] CHECKING AVAILABILITY OF RPM FUSION NVIDIA REPOSITORY...
[ ! ] RPM Fusion repository for Proprietary NVIDIA Driver was detected
[ ★ ] ATTEMPTING CONNECTION TO RPM FUSION SERVERS...
[ ✓ ] Connection to RPM Fusion servers was established
[ ★ ] LOOKING FOR EXISTING DRIVER PACKAGES...
      xorg-x11-drv-nvidia-kmodsrc-495.46-1.fc35.x86_64
      nvidia-libXNVCtrl-495.29.05-1.fc34.x86_64
      nvidia-driver-libs-495.29.05-1.fc34.x86_64
      nvidia-driver-495.29.05-1.fc34.x86_64
      nvidia-kmod-common-495.29.05-1.fc34.noarch
      kmod-nvidia-latest-dkms-495.29.05-1.fc34.x86_64
      akmod-nvidia-470.74-1.fc35.x86_64
      nvidia-settings-495.29.05-1.fc34.x86_64
      nvidia-driver-libs-495.29.05-1.fc34.i686
      xorg-x11-drv-nvidia-cuda-libs-495.46-1.fc35.x86_64
[ ! ] A total of 10 driver packages were detected
[ ★ ] REINSTALLING PROPRIETARY DRIVERS...
keybase                                                                                                      56 kB/s | 3.3 kB     00:00    
Das Paket gcc-11.2.1-7.fc35.x86_64 ist bereits installiert.
Das Paket kernel-headers-5.15.4-200.fc35.x86_64 ist bereits installiert.
Das Paket kernel-devel-5.15.10-200.fc35.x86_64 ist bereits installiert.
Das Paket kernel-devel-5.15.11-200.fc35.x86_64 ist bereits installiert.
Das Paket kernel-devel-5.15.12-200.fc35.x86_64 ist bereits installiert.
Das Paket akmod-nvidia-3:470.74-1.fc35.x86_64 ist bereits installiert.
Alle Übereinstimmungen wurden durch modulare Filterung nach Argumenten herausgefiltert: xorg-x11-drv-nvidia
Das Paket nvidia-driver-libs-3:495.29.05-1.fc34.x86_64 ist bereits installiert.
Das Paket nvidia-driver-libs-3:495.29.05-1.fc34.i686 ist bereits installiert.
Fehler: Es konnte kein Treffer gefunden werden.: xorg-x11-drv-nvidia
[ ✗ ] Proprietary drivers could not be installed
[ ✗ ] Leaving installer
```

For me, it looks like the problem being trigger by `xorg-x11-drv-nvidia-3:495.46-1.fc35.x86_64` being filtered out:

```bash
$ uname -a
Linux redsnapper 5.15.10-200.fc35.x86_64 #1 SMP Fri Dec 17 14:46:39 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
[tpasch@redsnapper ~]$ sudo dnf install akmod-nvidia
keybase                                                                                                      47 kB/s | 3.3 kB     00:00    
Das Paket akmod-nvidia-3:470.74-1.fc35.x86_64 ist bereits installiert.
Abhängigkeiten sind aufgelöst.

 Problem: package akmod-nvidia-3:495.46-1.fc35.x86_64 requires nvidia-kmod-common >= 3:495.46, but none of the providers can be installed
  - cannot install the best candidate for the job
  - package xorg-x11-drv-nvidia-3:495.46-1.fc35.x86_64 is filtered out by modular filtering
============================================================================================================================================
 Package                       Architecture            Version                             Repository                                  Size
============================================================================================================================================
Pakete mit nicht auflösbaren Abhängigkeiten werden übersprungen:
 akmod-nvidia                  x86_64                  3:495.46-1.fc35                     rpmfusion-nonfree-updates                   26 k

Transaktionsübersicht
============================================================================================================================================
Überspringen  1 Paket

Nichts zu tun.
Fertig.
```

```bash
$ nvautoinstall --version
NVAutoInstall by Akashdeep Dhar <t0xic0der@fedoraproject.org>, version 0.4.0
```

t0xic0der, 2022-01-08:
Hi @aanno, here's a support article that I wrote about this issue https://ask.fedoraproject.org/t/solution-modular-filtering-issue-for-nvidia-drivers-cuda-sources-conflict/14063. Please check if this helps solving your issue.

aanno, 2022-01-09:
Well, you suggestion (see below) works and also fixes your  `nvidia-auto-installer-for-fedora-linux`. However, plasma on X11 still not works with kernel 5.15.12 (but plasma on wayland is usable).

So, this is what I did:

```bash
# dnf module disable nvidia-driver
# dnf install --best --allowerasing  akmod-nvidia
```

Complete output for reference:
```bash
# dnf module disable nvidia-driver
keybase                                                                                                     8.2 kB/s | 3.3 kB     00:00    
Abhängigkeiten sind aufgelöst.
============================================================================================================================================
 Package                          Architecture                    Version                            Repository                        Size
============================================================================================================================================
Disabling modules:
 nvidia-driver                                                                                                                             

Transaktionsübersicht
============================================================================================================================================

Ist dies in Ordnung? [j/N]: j
Fertig.

# dnf install --best --allowerasing  akmod-nvidia
Letzte Prüfung auf abgelaufene Metadaten: vor 0:00:50 am So 09 Jan 2022 12:46:51 CET.
Das Paket akmod-nvidia-3:470.74-1.fc35.x86_64 ist bereits installiert.
Abhängigkeiten sind aufgelöst.
============================================================================================================================================
 Package                                Architecture         Version                          Repository                               Size
============================================================================================================================================
Aktualisieren:
 akmod-nvidia                           x86_64               3:495.46-1.fc35                  rpmfusion-nonfree-updates                26 k
 nvidia-settings                        x86_64               3:495.46-1.fc35                  rpmfusion-nonfree-updates               1.6 M
Abhängigkeiten werden installiert:
 xorg-x11-drv-nvidia                    x86_64               3:495.46-1.fc35                  rpmfusion-nonfree-updates                21 M
 xorg-x11-drv-nvidia-libs               i686                 3:495.46-1.fc35                  rpmfusion-nonfree-updates                24 M
 xorg-x11-drv-nvidia-libs               x86_64               3:495.46-1.fc35                  rpmfusion-nonfree-updates               149 M
Abhängige Pakete werden entfernt:
 kmod-nvidia-latest-dkms                x86_64               3:495.29.05-1.fc34               @cuda-fedora34-x86_64                    48 M
 nvidia-driver                          x86_64               3:495.29.05-1.fc34               @cuda-fedora34-x86_64                    56 M
 nvidia-driver-libs                     i686                 3:495.29.05-1.fc34               @cuda-fedora34-x86_64                    77 M
 nvidia-driver-libs                     x86_64               3:495.29.05-1.fc34               @cuda-fedora34-x86_64                   334 M
 nvidia-kmod-common                     noarch               3:495.29.05-1.fc34               @cuda-fedora34-x86_64                   1.8 k

Transaktionsübersicht
============================================================================================================================================
Installieren   3 Pakete
Aktualisieren  2 Pakete
Entfernen      5 Pakete

Gesamte Downloadgröße: 195 M
Pakete werden heruntergeladen:
(1/5): xorg-x11-drv-nvidia-495.46-1.fc35.x86_64.rpm                                                         2.3 MB/s |  21 MB     00:08    
(2/5): akmod-nvidia-495.46-1.fc35.x86_64.rpm                                                                 86 kB/s |  26 kB     00:00    
(3/5): nvidia-settings-495.46-1.fc35.x86_64.rpm                                                             1.7 MB/s | 1.6 MB     00:00    
(4/5): xorg-x11-drv-nvidia-libs-495.46-1.fc35.i686.rpm                                                      2.1 MB/s |  24 MB     00:11    
(5/5): xorg-x11-drv-nvidia-libs-495.46-1.fc35.x86_64.rpm                                                    4.3 MB/s | 149 MB     00:34    
--------------------------------------------------------------------------------------------------------------------------------------------
Gesamt                                                                                                      5.5 MB/s | 195 MB     00:35     
Transaktionsüberprüfung wird ausgeführt
Transaktionsprüfung war erfolgreich.
Transaktion wird getestet
Transaktionstest war erfolgreich.
Transaktion wird ausgeführt
  Vorbereitung läuft    :                                                                                                               1/1 
  Installieren          : xorg-x11-drv-nvidia-libs-3:495.46-1.fc35.i686                                                                1/12 
  Installieren          : xorg-x11-drv-nvidia-libs-3:495.46-1.fc35.x86_64                                                              2/12 
[... 57 lines cut ...]
      xorg-x11-drv-nvidia-cuda-libs-495.46-1.fc35.x86_64
      xorg-x11-drv-nvidia-libs-495.46-1.fc35.i686
[... cut ...]
Das Paket kernel-devel-5.15.12-200.fc35.x86_64 ist bereits installiert.
Das Paket akmod-nvidia-3:495.46-1.fc35.x86_64 ist bereits installiert.
Das Paket xorg-x11-drv-nvidia-3:495.46-1.fc35.x86_64 ist bereits installiert.
Das Paket xorg-x11-drv-nvidia-libs-3:495.46-1.fc35.i686 ist bereits installiert.
Das Paket xorg-x11-drv-nvidia-libs-3:495.46-1.fc35.x86_64 ist bereits installiert.
Abhängigkeiten sind aufgelöst.
Nichts zu tun.
Fertig.
[ ✓ ] Driver package installation completed
[ ✗ ] Leaving installer
```

t0xic0der, 2022-01-09:
> However, plasma on X11 still not works with kernel 5.15.12 (but plasma on wayland is usable).

Unfortunately, this has to do with the driver and not the installer as the installer has apparently configured the driver properly. I would suggest reaching out to the folks at RPM Fusion who work on maintaining these drivers and see if they have any solutions for this.

aanno, 2022-01-11:
Reported the problem at https://bugzilla.rpmfusion.org/show_bug.cgi?id=6193
Thank you for suggestion!

aanno, 2022-01-11:
BTW, isn't `nvautoinstall` to solve issues like the one I encountered automatically?

t0xic0der, 2022-01-12:
@aanno, it should as I mentioned in https://ask.fedoraproject.org/t/solution-modular-filtering-issue-for-nvidia-drivers-cuda-sources-conflict/14063/2?u=t0xic0der but in the presence of conflicting packages, the tool would rather prefer to exit out with an error than to assume that a certain package is not important and hence, remove it.

I have intentionally not included the directives that allow for erasing of existing conflicting packages as well as repositories in order to install the drivers.

aanno, 2022-01-16:
I reported the x11 problem at https://forums.developer.nvidia.com/t/x11-has-stopped-working-on-fedora-35/200682 (like suggested from rpmfusion).

aanno, 2022-01-16:
Button line: 
* On `01:00.0 VGA compatible controller: NVIDIA Corporation TU106 [GeForce RTX 2060 Rev. A] (rev a1)` the rpmfusion package `xorg-x11-drv-nvidia` version `3:495.46-1.fc35` does _not_ work any more
* On this card you need `NVIDIA-Linux-x86_64-470.94.run` nowerdays (`NVIDIA-Linux-x86_64-495.46.run`) does _not_ work.

aanno, 2022-01-17:
Because of the problem mentioned above, `nvautoinstall driver` will trash nvidia x11 support (as `NVIDIA-Linux-x86_64-470.94.run` is installed and will get overwritten).

```bash
$ sudo nvautoinstall  driver
[ # ] NVIDIA AUTOINSTALLER FOR FEDORA LINUX
[ ★ ] CHECKING SUPERUSER PERMISSIONS...
[ ✓ ] Superuser privilege acquired
[ ★ ] CHECKING AVAILABILITY OF RPM FUSION NVIDIA REPOSITORY...
[ ! ] RPM Fusion repository for Proprietary NVIDIA Driver was detected
[ ★ ] ATTEMPTING CONNECTION TO RPM FUSION SERVERS...
[ ✓ ] Connection to RPM Fusion servers was established
[ ★ ] LOOKING FOR EXISTING DRIVER PACKAGES...
      nvidia-libXNVCtrl-495.29.05-1.fc34.x86_64
      xorg-x11-drv-nvidia-cuda-libs-495.46-1.fc35.x86_64
[ ! ] A total of 2 driver packages were detected
[ ★ ] REINSTALLING PROPRIETARY DRIVERS...
Schnellster Mirror wird ermittelt (5 Hosts).. done.    [===                                               ] ---  B/s |   0  B     --:-- ETA
Fedora 35 - x86_64 - Updates                                                                                 12 kB/s |  13 kB     00:01    
done.a 35 - x86_64 - Updates                           [   ===                                            ] ---  B/s |   0  B     --:-- ETA
Fedora 35 - x86_64 - Updates                                                                                1.3 MB/s | 2.1 MB     00:01    
Fedora Modular 35 - x86_64 - Updates                                                                         81 kB/s |  13 kB     00:00    
keybase                                                                                                      11 kB/s | 3.3 kB     00:00    
Das Paket gcc-11.2.1-7.fc35.x86_64 ist bereits installiert.
Das Paket kernel-headers-5.15.4-200.fc35.x86_64 ist bereits installiert.
Das Paket kernel-devel-5.15.12-200.fc35.x86_64 ist bereits installiert.
Das Paket kernel-devel-5.15.13-200.fc35.x86_64 ist bereits installiert.
Das Paket kernel-devel-5.15.14-200.fc35.x86_64 ist bereits installiert.
Abhängigkeiten sind aufgelöst.
============================================================================================================================================
 Package                                   Architecture         Version                       Repository                               Size
============================================================================================================================================
Installieren:
 akmod-nvidia                              x86_64               3:495.46-1.fc35               rpmfusion-nonfree-updates                26 k
Abhängigkeiten werden installiert:
 egl-wayland                               x86_64               1.1.9-3.fc35                  updates                                  36 k
 libglvnd-gles                             i686                 1:1.3.4-2.fc35                updates                                  30 k
 libglvnd-opengl                           i686                 1:1.3.4-2.fc35                updates                                  40 k
 nvidia-settings                           x86_64               3:495.46-1.fc35               rpmfusion-nonfree-updates               1.6 M
 xorg-x11-drv-nvidia                       x86_64               3:495.46-1.fc35               rpmfusion-nonfree-updates                21 M
 xorg-x11-drv-nvidia-kmodsrc               x86_64               3:495.46-1.fc35               rpmfusion-nonfree-updates                26 M
 xorg-x11-drv-nvidia-libs                  i686                 3:495.46-1.fc35               rpmfusion-nonfree-updates                24 M
 xorg-x11-drv-nvidia-libs                  x86_64               3:495.46-1.fc35               rpmfusion-nonfree-updates               149 M

Transaktionsübersicht
============================================================================================================================================
Installieren  9 Pakete
```

Perhaps `nvautoinstaller` should ask if you really want to overwrite the proprietary nvidia drivers package.

t0xic0der, 2022-01-18:
> Because of the problem mentioned above, nvautoinstall driver will trash nvidia x11 support (as NVIDIA-Linux-x86_64-470.94.run is installed and will get overwritten).

Please refer to this https://ask.fedoraproject.org/t/xorg-x11-drv-nouveau-error-and-nvidia-kernel-problem/13827/12?u=t0xic0der. We do not recommend the use of RUN file due to the aforementioned reasons and that is why the installer is intentionally made to be oblivious of its existence.