gridhead/nvidia-auto-installer-for-fedora-linux #99: Unable to find a match: kernel-headers kernel-devel

sugamkarki, 2022-05-02:
```
[ftickler@2400-1A00-B020 ~]$ sudo nvautoinstall driver
[ # ] NVIDIA AUTOINSTALLER FOR FEDORA LINUX
[ ★ ] CHECKING SUPERUSER PERMISSIONS...
[ ✓ ] Superuser privilege acquired
[ ★ ] CHECKING AVAILABILITY OF RPM FUSION NVIDIA REPOSITORY...
[ ! ] RPM Fusion repository for Proprietary NVIDIA Driver was detected
[ ★ ] ATTEMPTING CONNECTION TO RPM FUSION SERVERS...
[ ✓ ] Connection to RPM Fusion servers was established
[ ★ ] LOOKING FOR EXISTING DRIVER PACKAGES...
[ ! ] No existing NVIDIA driver packages were detected
[ ★ ] INSTALLING PROPRIETARY DRIVERS...
Last metadata expiration check: 0:02:40 ago on Mon 02 May 2022 01:20:38 PM +0545.
Package gcc-11.3.1-2.fc35.x86_64 is already installed.
All matches were filtered out by exclude filtering for argument: kernel-headers
All matches were filtered out by exclude filtering for argument: kernel-devel
Error: Unable to find a match: kernel-headers kernel-devel
[ ✗ ] Proprietary drivers could not be installed
[ ✗ ] Leaving installer

```

My system:
```
OS: Fedora Linux 35 (KDE Plasma) x86_64 
Kernel: 5.14.10-300.fc35.x86_64 
```

t0xic0der, 2022-05-03:
Hi @sugamkarki,

Do you happen to be facing this problem even now or is it fixed?

sugamkarki, 2022-05-03:
Yes it is solved. I had to upgrade me kernel to the latest to fix this. Thank you!

t0xic0der, 2022-05-03:
Glad to know. I should ensure that all packages are updated to the their latest version before I start with the installation so I would open this ticket up. Please note that this is not a bug but more of a Quality-of-Life improvement to the installation implementation.