gridhead/nvidia-auto-installer-for-fedora-linux #107: Create an option to generate cert on secure boot

MysterHawk, 2022-05-24:
Hi!

I've noticed a bug, the program is almost perfect but if I can add one thing, if you install the nvidia drivers and you have secure boot enabled, you will not be able to load nvidia kernel modules on boot, you have to sign them with a self generated certs that prove that those modules can be trusted.

This is the phrase that I would get on boot:
nvidia kernel module missing falling back to nouveau

I followed these steps to resolve the issue:
1. sudo dnf remove kmod-nvidia-5.17.8* (the actual  version for the kernel)
2. sudo kmodgenca
3. sudo mokutil --import /etc/pki/akmods/certs/public_key.der
4. sudo akmods --force (Then wait about 3-5 minutes)
5. dnf list installed kmod-nvidia* should now show a new kmod-nvidia for the current kernel 5.17.4 (if not it will be created during the next boot)
6. reboot and repeat the import step during boot.
7. Now it should boot and load the driver since the new driver will be signed by the latest key generated.

(the steps are taken from this answer https://ask.fedoraproject.org/t/installing-nvidia-drivers-on-a-fedora-36-in-dual-boot-with-windows-11-with-secure-boot-enabled/21532)

It would be nice to have implemented this feature :)

t0xic0der, 2022-05-26:
@MysterHawk,

Thank you for reaching out with a feature request. Unfortunately, these steps are, to a great extent, subjective to the kernel which is installed on the user's device at a certain point in time - which would mean that these need to be run every after a kernel update as the secure boot signing would break in the absence of the older kernel where the sign was available.

As much as I think that implementing this feature is well within the scope of the project, I also believe that this would lead to users making undue modifications to their system that they probably would have no idea about as this tool is targetted towards those who want to get things done quickly without looking into the technical aspect of it all.

Technically-abled users should be able to find their way to mentioned guide on the RPM Fusion docs (i.e. https://rpmfusion.org/Howto/Secure%20Boot) and follow through the process by themselves. At the same time, I like to think that the users who are not very tech-savvy would not be bothered much by having the secure boot turned off, in my honest opinion.

I am curious to know what you think.

MysterHawk, 2022-06-15:
Sorry for the late answer,

It is true that this feature request is a bit more advanced that the others functionality of your application, but in a UEFI system your method to install nvidia drivers won't work, the users won't be able to understand withouth a bit of research what happend on the first boot (kernel modules not loaded...), I think that atleast for the UEFI user a warning is needed when the nvidia auto installer is launched, to explain briefly what needs to be done to make the uefi installation work, in my opinion after that an option to automate this would be pretty neet :)

You only need to to this onetime, after that the system will sign automatically the driver updated for what I understand.

nowadays have the secureboot enabled is a must unfortunately, if you are using windows 11 in dualboot windows will require it to work properly.

onuralpszr, 2022-12-19:
@t0xic0der  today secureboot option is works fine and we can present 2 options to users 

1. We can ask for (Do you have cert file and If yes we can ask for path and file names
2. We can generate based on the document rpmfusion present and we can say that to user "you will see "blue screen" for approve the cer and make sure It is signed. (in that process WE CAN create "new systemd" file (temp) and make sure it is active and base on that option we can give confidence to user to say that "your driver signed and works fine" ( we can check lsmod and check cert loaded (there are command line tools for that ) 

THE reason I want this because "It is better sign your PC first and install next" otherwise you have to extra steps so that is why I think we need to make sure it is there and works fine" I tested a lot of times. Having sign first better because "other kmod drivers" also automatically works as well.

onuralpszr, 2023-01-14:
> - which would mean that these need to be run every after a kernel update as the secure boot signing would break in the absence of the older kernel where the sign was available.

No they don't (akmod keeps them updated) so it is VERY EASY