gridhead/nvidia-auto-installer-for-fedora-linux #110: Blackscreen after usage.

QazCetelic, 2022-07-07:
I've used the tool to install the drivers but sadly this has caused an issue. The login screen does not show up. I managed to click on the password field by pure luck and was able to log in. Than the black screen was replaced with the startup animation after which it did function. The GPU is now automatically used when launching certain applications, but they do crash.

QazCetelic, 2022-07-07:
System information:
Operating System: Fedora Linux 36
KDE Plasma Version: 5.24.5
KDE Frameworks Version: 5.94.0
Qt Version: 5.15.3
Kernel Version: 5.18.9-200.fc36.x86_64 (64-bit)
Graphics Platform: X11
Processors: 12 × Intel® Core™ i7-8750H CPU @ 2.20GHz
Memory: 15.4 GiB of RAM
Graphics Processor: NVIDIA GeForce GTX 1060/PCIe/SSE2

QazCetelic, 2022-07-07:
I started experiencing these issues after using `sudo nvautoinstall primec`.

QazCetelic, 2022-07-07:
Output of `sudo nvautoinstall compat`:
```
[ # ] NVIDIA AUTOINSTALLER FOR FEDORA LINUX
[ ★ ] CHECKING FOR GPU COMPATIBILITY...
[ ! ] Compatibility infomation was obtained
[ ✓ ] One or more active NVIDIA GPUs were detected
      00:02.0 VGA compatible controller: Intel Corporation CoffeeLake-H GT2 [UHD Graphics 630]
      01:00.0 VGA compatible controller: NVIDIA Corporation GP106M [GeForce GTX 1060 Mobile] (rev a1)
[ ✓ ] An Optimus Dual GPU setup was detected
[ ★ ] GATHERING CURRENT HOST INFORMATION...
[ ! ] Host information was gathered
      System: Linux v5.18.9-200.fc36.x86_64
      Hostname: fedora
      Version: #1 SMP PREEMPT_DYNAMIC Sat Jul 2 15:56:43 UTC 2022
      Distribution: Fedora Linux x86_64
[ ★ ] CHECKING FOR HOST COMPATIBILITY...
[ ✓ ] Supported OS detected
      This tool is expected to work correctly here
[ ✗ ] Leaving installer
```

QazCetelic, 2022-07-07:
Reverting the NVIDIA driver install with the RPMfusion guide fixed the SDDM and applications.

QazCetelic, 2022-07-07:
I managed to get it to start after the 4th reboot (??) and Nvidia X-Server settings works now. GreenWithEnvy is also able to get data from the GPU. Applications do start now, but use the iGPU.

QazCetelic, 2022-07-07:
I have been able to reproduce the same issue (blackscreen SDDM & broken apps) when manually setting the Nvidia GPU as primary GPU with the [fedora docs guide](https://docs.fedoraproject.org/en-US/quick-docs/how-to-set-nvidia-as-primary-gpu-on-optimus-based-laptops/#_why_would_you_want_to_do_that).

QazCetelic, 2022-07-07:
I have reverted the change by removing the `Option "PrimaryGPU" "yes"` line in `/etc/X11/xorg.conf.d/nvidia.conf`. The SDDM and applications are functioning again.

kphanipavan, 2022-07-08:
Have you tried changing /etc/sddm/Xsetup file?

If not, refer the `note` in the below link.
https://docs.fedoraproject.org/en-US/quick-docs/how-to-set-nvidia-as-primary-gpu-on-optimus-based-laptops/#_step_8_edit_the_x11_configuration

Change /etc/sddm/Xsetup file to contain the following:
```
xrandr --setprovideroutputsource modesetting NVIDIA-G0
xrandr --auto
```

I have a similar setup, an Intel CPU and an Nvidia GPU. Prime and reverse Prime works with the above Xsetup mod.

Dynamic power boost and D3 power management is working fine too. Can you report on that as well please...

QazCetelic, 2022-07-08:
This works, the SDDM is properly shown now, but the some applications still crash.

QazCetelic, 2022-07-08:
It seems like Nvidia is the primary GPU on my system now. Apps crash with and without the wrapper command; `env __NV_PRIME_RENDER_OFFLOAD=1 env __GLX_VENDOR_LIBRARY_NAME=nvidia`.

[image]

kphanipavan, 2022-07-08:
Switch to nvidia gpu being the primary gpu, and dont use the env flags...

QazCetelic, 2022-07-08:
I think I've already chosen Nvidia to be my primary GPU.
`nvidia.conf`:
```
#This file is provided by xorg-x11-drv-nvidia
#Do not edit

Section "OutputClass"
	Identifier "nvidia"
	MatchDriver "nvidia-drm"
	Driver "nvidia"
	Option "AllowEmptyInitialConfiguration"
	Option "SLI" "Auto"
	Option "BaseMosaic" "on"
	Option "PrimaryGPU" "yes"
EndSection

Section "ServerLayout"
	Identifier "layout"
	Option "AllowNVIDIAGPUScreens"
EndSection
```

I'm trying to use the [PolyMC launcher](https://polymc.org/) (Minecraft Launcher).
I get the following error when using your suggested wrapper command `The wrapper command "__GLX_VENDOR_LIBRARY_NAME=nvidia" couldn't be found.`.

kphanipavan, 2022-07-08:
With Nvidia GPU as primary, dont use that wrapper, launch the app directly.

QazCetelic, 2022-07-08:
The apps work using the RPM install.