gridhead/nvidia-auto-installer-for-fedora-linux #117: Conflicting requests when install nvautoinstall in FEDORA 37

hhlp, 2022-09-22:
``` 
sudo dnf install nvautoinstall   
```    

``` 
Error: 
 Problem: conflicting requests
  - nothing provides python(abi) = 3.10 needed by nvautoinstall-0.4.1-0.fc37.noarch
  - nothing provides python3.10dist(click) >= 7.1.2 needed by nvautoinstall-0.4.1-0.fc37.noarch
  - nothing provides python3.10dist(distro) >= 1.5 needed by nvautoinstall-0.4.1-0.fc37.noarch
(try to add '--skip-broken' to skip uninstallable packages)
```

Regards.,

t0xic0der, 2022-09-29:
@hhlp,

Thanks for reporting. I will take a look.

closed 2022-10-09 by t0xic0der

onuralpszr, 2022-10-09:
After build of copr repo It works fine. I do also have python 3.11 and python 3.10 at the same time and all works here. 

```
 dnf install nvautoinstall -y

Installing:
 nvautoinstall noarch 0.4.1-2.fc37  copr:copr.fedorainfracloud.org:t0xic0der:nvidia-auto-installer-for-fedora  56 k

Install  1 Package

Total download size: 56 k
Installed size: 239 k
Downloading Packages:
nvautoinstall-0.4.1-2.fc37.noarch.rpm                                               104 kB/s |  56 kB     00:00    
--------------------------------------------------------------------------------------------------------------------
Total                                                                               103 kB/s |  56 kB     00:00     
Running transaction check
Transaction check succeeded.
Running transaction test
Transaction test succeeded.
Running transaction
  Preparing        :                                                                                            1/1 
  Installing       : nvautoinstall-0.4.1-2.fc37.noarch                                                          1/1 
  Running scriptlet: nvautoinstall-0.4.1-2.fc37.noarch                                                          1/1 
  Verifying        : nvautoinstall-0.4.1-2.fc37.noarch                                                          1/1 

Installed:
  nvautoinstall-0.4.1-2.fc37.noarch                   
```

```
nvautoinstall --version
NVAutoInstall by Akashdeep Dhar <t0xic0der@fedoraproject.org>, version 0.4.1
```

```
nvautoinstall compat
[ # ] NVIDIA AUTOINSTALLER FOR FEDORA LINUX
[ ★ ] CHECKING FOR GPU COMPATIBILITY...
[ ! ] Compatibility infomation was obtained
[ ✓ ] One or more active NVIDIA GPUs were detected
      ........... (deleted)
[ ✓ ] An Optimus Dual GPU setup was detected
[ ★ ] GATHERING CURRENT HOST INFORMATION...
[ ! ] Host information was gathered
      System: Linux v5.19.14-300.fc37.x86_64
      Hostname: localhost
      Version: #1 SMP PREEMPT_DYNAMIC Wed Oct 5 21:31:43 UTC 2022
      Distribution: Fedora Linux x86_64
[ ★ ] CHECKING FOR HOST COMPATIBILITY...
[ ✓ ] Supported OS detected
      This tool is expected to work correctly here
[ ✗ ] Leaving installer

```