gridhead/nvidia-auto-installer-for-fedora-linux #127: Nvidia-vaapi-driver installation and  how to manage within the script :construction:

onuralpszr, 2022-10-10:
Rpmfusion folks already add this package into the repository for having nvidia-vaapi but we need strategy how do we make it easier for users. (It needs some config changes some env params)  

Source of project : https://github.com/elFarto/nvidia-vaapi-driver

Ideas I was think of ;

- Having a custom desktop file for firefox (If user wants to)
- Set those firefox config accordingly (If user wants to)
- User needs to understand that it will delete libva-vdpau package because of the symlink comes with nvidia-vaapi-driver

Snippet from rpm spec file

```

# The nvidia_drv_video.so symlink is also provided by theses packages
Conflicts: libva-vdpau-driver%{?_isa}
Conflicts: libva-vdpau-driver-vp9%{?_isa}

```

t0xic0der, 2022-12-22:
I am curious to know what our friendly resident VAAPI expert @akarshanbiswas thinks of this.

akarshanbiswas, 2022-12-22:
@t0xic0der Thank you for bringing it to my attention. 

In my opinion it can be done via `sudo nvautoinstall vidacc --experimental` or something. Instead of setting desktop files, we have to only set environment variables.

setting the configs via script will not be a very good idea (even if the user wants it, because reverting back will be tedious in my opinion), so we need an escape hatch. 

To be honest, the upcoming Vulkan hardware accelerated Video decode (atleast for h264 and h265) will be soon be a reality and will solve all these problems (only if Mozilla gets serious and reincarnate it's dying webrender-vulkan fork) :)