gridhead/nvidia-auto-installer-for-fedora-linux #153: CuDNN and CuBLAS auto-install on Fedora 37

bstee615, 2022-12-21:
Thank you for releasing this very useful tool -- it was the only way I could successfully install CUDA for PyTorch on my fresh Fedora 37 installation. However, there were some manual steps which I needed to take after running the tool. It would be awesome if these steps could be automated as part of the tool.

1. libcublas.so.11 was missing. I installed the library using `sudo dnf install libcublas-11-8`.
2. libcudnn.so.8 was missing. I installed the library manually, by following these instructions plus adding `/usr/local/cuda/lib64` to my library path using `ldconfig`: https://docs.nvidia.com/deeplearning/cudnn/install-guide/index.html#installlinux-tar.

I imagine both of these steps could be automated fairly easily as new steps in this tool, such as `nvautoinstall cublas` and `nvautoinstall cudnn`. There is a small issue of the login required for cudnn.

This addition would allow users to install all requisite libraries using only this tool, and would allow users to install other auxiliary libraries that are not included in the base CUDA installation. I'm willing to try making a PR for it if you think it will be a useful addition, though it will probably take a number of weeks for me to implement it.

t0xic0der, 2022-12-22:
@bstee615, thanks for suggesting the feature. 

@onuralpszr, does this feature look like something that could be of help for the users?

onuralpszr, 2022-12-22:
> @bstee615, thanks for suggesting the feature.
> 
> @onuralpszr, does this feature look like something that could be of help for the users?

We had a similar discussion before but, we have to make sure also we need to install proper cuda as well  for example cublas-11.8 need cuda 11.8 so we have to install that too. I am checking the PR we may need to do some adjustment and I want to test as well in "pure pytorch" installation and after we found safe solution we can try to extend into other AI/ML frameworks such as "onnx" / "tensorflow" / "ncnn"

onuralpszr, 2022-12-22:
> 2\. libcudnn.so.8 was missing. I installed the library manually, by following these instructions plus adding `/usr/local/cuda/lib64` to my library path using `ldconfig`: https://docs.nvidia.com/deeplearning/cudnn/install-guide/index.html#installlinux-tar.

There is a RPM package for that, we don't have to do this.

onuralpszr, 2022-12-22:
For skip login side in cudnn 

https://developer.download.nvidia.com/compute/redist/cudnn/v8.7.0/local_installers/11.8/

No login required

bstee615, 2022-12-24:
> For skip login side in cudnn 
> 
> 
> 
> https://developer.download.nvidia.com/compute/redist/cudnn/v8.7.0/local_installers/11.8/
> 
> 
> 
> No login required 

This is a great find, it's a lot better than using RHEL8 repos. Thanks