gridhead/nvidia-auto-installer-for-fedora-linux #160: not install in rawhide ( Error: Unable to find a match: xorg-x11-drv-nvidia xorg-x11-drv-nvidia-libs )

xxavi-hola-mon, 2023-01-01:
Hi ,

As you can see, I get an error. Please, if you @t0xic0der  need more logs, tell me !

[root@fedora ~]# 
[root@fedora ~]# date
Sun Jan  1 04:50:48 AM CET 2023
[root@fedora ~]# 
[root@fedora ~]# uname -a
Linux fedora 6.2.0-0.rc1.20221230gitbff687b3dad6.15.fc38.x86_64 #1 SMP PREEMPT_DYNAMIC Fri Dec 30 14:23:36 UTC 2022 x86_64 GNU/Linux
[root@fedora ~]# 
[root@fedora ~]# nvautoinstall driver
[ # ] NVIDIA AUTOINSTALLER FOR FEDORA LINUX
[ ★ ] CHECKING SUPERUSER PERMISSIONS...
[ ✓ ] Superuser privilege acquired
[ ★ ] CHECKING AVAILABILITY OF RPM FUSION NVIDIA REPOSITORY...
[ ! ] RPM Fusion repository for Proprietary NVIDIA Driver was detected
[ ★ ] ATTEMPTING CONNECTION TO RPM FUSION SERVERS...
[ ✓ ] Connection to RPM Fusion servers was established
[ ★ ] LOOKING FOR EXISTING DRIVER PACKAGES...
      error: rpmdbNextIterator: skipping h#   12714 
      Header V4 DSA/SHA1 Signature, key ID 7fac5991: BAD
      Header SHA256 digest: OK
      Header SHA1 digest: OK
      error: rpmdbNextIterator: skipping h#   12715 
      Header V4 DSA/SHA1 Signature, key ID 7fac5991: BAD
      Header SHA256 digest: OK
      Header SHA1 digest: OK
      error: rpmdbNextIterator: skipping h#   12716 
      Header V4 DSA/SHA1 Signature, key ID 7fac5991: BAD
      Header SHA256 digest: OK
      Header SHA1 digest: OK
      xorg-x11-drv-nvidia-kmodsrc-525.60.11-1.fc38.x86_64
      xorg-x11-drv-nvidia-cuda-libs-525.60.11-1.fc38.x86_64
      xorg-x11-drv-nvidia-libs-525.60.11-1.fc38.i686
      xorg-x11-drv-nvidia-libs-525.60.11-1.fc38.x86_64
      xorg-x11-drv-nvidia-power-525.60.11-1.fc38.x86_64
      xorg-x11-drv-nvidia-525.60.11-1.fc38.x86_64
      xorg-x11-drv-nvidia-cuda-libs-525.60.11-1.fc38.i686
      xorg-x11-drv-nvidia-cuda-525.60.11-1.fc38.x86_64
      nvidia-settings-525.60.11-1.fc38.x86_64
      nvidia-persistenced-525.60.11-1.fc38.x86_64
      akmod-nvidia-525.60.11-1.fc38.x86_64
      pcp-pmda-nvidia-gpu-6.0.1-4.fc38.x86_64
      nvidia-gpu-firmware-20221214-145.fc38.noarch
      nvidia-xconfig-debugsource-525.60.11-1.fc38.x86_64
      nvidia-xconfig-debuginfo-525.60.11-1.fc38.x86_64
      nvidia-xconfig-525.60.11-1.fc38.x86_64
[ ! ] A total of 28 driver packages were detected
[ ★ ] REINSTALLING PROPRIETARY DRIVERS...
Last metadata expiration check: 0:17:14 ago on Sun 01 Jan 2023 04:33:53 AM CET.
Package gcc-12.2.1-4.fc38.x86_64 is already installed.
Package kernel-headers-6.1.0-1.fc38.x86_64 is already installed.
Package kernel-devel-6.2.0-0.rc0.20221219gitf9ff5644bcc0.7.fc38.x86_64 is already installed.
Package kernel-devel-6.2.0-0.rc1.14.fc38.x86_64 is already installed.
Package kernel-devel-6.2.0-0.rc1.20221230gitbff687b3dad6.15.fc38.x86_64 is already installed.
Package akmod-nvidia-3:525.60.11-1.fc38.x86_64 is already installed.
All matches were filtered out by exclude filtering for argument: xorg-x11-drv-nvidia
All matches were filtered out by exclude filtering for argument: xorg-x11-drv-nvidia-libs
Error: Unable to find a match: xorg-x11-drv-nvidia xorg-x11-drv-nvidia-libs
[ ✗ ] Proprietary drivers could not be installed
      Please try executing `dnf update` with elevated privileges before this
[ ✗ ] Leaving installer
[root@fedora ~]# 
[root@fedora ~]# dnf update
Last metadata expiration check: 0:17:22 ago on Sun 01 Jan 2023 04:33:53 AM CET.
Dependencies resolved.
Nothing to do.
Complete!
[root@fedora ~]# 
[root@fedora ~]# nvautoinstall driver
[ # ] NVIDIA AUTOINSTALLER FOR FEDORA LINUX
[ ★ ] CHECKING SUPERUSER PERMISSIONS...
[ ✓ ] Superuser privilege acquired
[ ★ ] CHECKING AVAILABILITY OF RPM FUSION NVIDIA REPOSITORY...
[ ! ] RPM Fusion repository for Proprietary NVIDIA Driver was detected
[ ★ ] ATTEMPTING CONNECTION TO RPM FUSION SERVERS...
[ ✓ ] Connection to RPM Fusion servers was established
[ ★ ] LOOKING FOR EXISTING DRIVER PACKAGES...
      error: rpmdbNextIterator: skipping h#   12714 
      Header V4 DSA/SHA1 Signature, key ID 7fac5991: BAD
      Header SHA256 digest: OK
      Header SHA1 digest: OK
      error: rpmdbNextIterator: skipping h#   12715 
      Header V4 DSA/SHA1 Signature, key ID 7fac5991: BAD
      Header SHA256 digest: OK
      Header SHA1 digest: OK
      error: rpmdbNextIterator: skipping h#   12716 
      Header V4 DSA/SHA1 Signature, key ID 7fac5991: BAD
      Header SHA256 digest: OK
      Header SHA1 digest: OK
      xorg-x11-drv-nvidia-kmodsrc-525.60.11-1.fc38.x86_64
      xorg-x11-drv-nvidia-cuda-libs-525.60.11-1.fc38.x86_64
      xorg-x11-drv-nvidia-libs-525.60.11-1.fc38.i686
      xorg-x11-drv-nvidia-libs-525.60.11-1.fc38.x86_64
      xorg-x11-drv-nvidia-power-525.60.11-1.fc38.x86_64
      xorg-x11-drv-nvidia-525.60.11-1.fc38.x86_64
      xorg-x11-drv-nvidia-cuda-libs-525.60.11-1.fc38.i686
      xorg-x11-drv-nvidia-cuda-525.60.11-1.fc38.x86_64
      nvidia-settings-525.60.11-1.fc38.x86_64
      nvidia-persistenced-525.60.11-1.fc38.x86_64
      akmod-nvidia-525.60.11-1.fc38.x86_64
      pcp-pmda-nvidia-gpu-6.0.1-4.fc38.x86_64
      nvidia-gpu-firmware-20221214-145.fc38.noarch
      nvidia-xconfig-debugsource-525.60.11-1.fc38.x86_64
      nvidia-xconfig-debuginfo-525.60.11-1.fc38.x86_64
      nvidia-xconfig-525.60.11-1.fc38.x86_64
[ ! ] A total of 28 driver packages were detected
[ ★ ] REINSTALLING PROPRIETARY DRIVERS...
Last metadata expiration check: 0:17:36 ago on Sun 01 Jan 2023 04:33:53 AM CET.
Package gcc-12.2.1-4.fc38.x86_64 is already installed.
Package kernel-headers-6.1.0-1.fc38.x86_64 is already installed.
Package kernel-devel-6.2.0-0.rc0.20221219gitf9ff5644bcc0.7.fc38.x86_64 is already installed.
Package kernel-devel-6.2.0-0.rc1.14.fc38.x86_64 is already installed.
Package kernel-devel-6.2.0-0.rc1.20221230gitbff687b3dad6.15.fc38.x86_64 is already installed.
Package akmod-nvidia-3:525.60.11-1.fc38.x86_64 is already installed.
All matches were filtered out by exclude filtering for argument: xorg-x11-drv-nvidia
All matches were filtered out by exclude filtering for argument: xorg-x11-drv-nvidia-libs
Error: Unable to find a match: xorg-x11-drv-nvidia xorg-x11-drv-nvidia-libs
[ ✗ ] Proprietary drivers could not be installed
      Please try executing `dnf update` with elevated privileges before this
[ ✗ ] Leaving installer
[root@fedora ~]# 
[root@fedora ~]#

onuralpszr, 2023-01-02:
For simple test sake could you do those (on my end it works and find those packages) 

```shell
dnf install https://mirrors.rpmfusion.org/free/fedora/rpmfusion-free-release-$(rpm -E %fedora).noarch.rpm https://mirrors.rpmfusion.org/nonfree/fedora/rpmfusion-nonfree-release-$(rpm -E %fedora).noarch.rpm

dnf install xorg-x11-drv-nvidia xorg-x11-drv-nvidia-libs
```

```
Installing:
 xorg-x11-drv-nvidia             x86_64 3:525.60.11-1.fc38                          rpmfusion-nonfree-rawhide  42 M
 ```

xxavi-hola-mon, 2023-01-08:
Hi @onuralpszr ,

I do a new fresh installation of Fedora rawhide and, it seems now was installed but, not loaded:

```

[root@localhost-live ~]# 
[root@localhost-live ~]# date
Sun Jan  8 02:53:02 AM CET 2023
[root@localhost-live ~]# 
[root@localhost-live ~]# dnf update
Last metadata expiration check: 0:02:28 ago on Sun 08 Jan 2023 02:50:41 AM CET.
Dependencies resolved.
Nothing to do.
Complete!
[root@localhost-live ~]# 
[root@localhost-live ~]# 
[root@localhost-live ~]# dnf install dnf-plugins-core -y
Last metadata expiration check: 0:02:36 ago on Sun 08 Jan 2023 02:50:41 AM CET.
Package dnf-plugins-core-4.3.1-1.fc38.noarch is already installed.
Dependencies resolved.
Nothing to do.
Complete!
[root@localhost-live ~]# 
[root@localhost-live ~]# 
[root@localhost-live ~]# dnf copr enable t0xic0der/nvidia-auto-installer-for-fedora -y
Enabling a Copr repository. Please note that this repository is not part
of the main distribution, and quality may vary.

The Fedora Project does not exercise any power over the contents of
this repository beyond the rules outlined in the Copr FAQ at
<https://docs.pagure.org/copr.copr/user_documentation.html#what-i-can-build-in-copr>,
and packages are not held to any quality or security level.

Please do not file bug reports about these packages in Fedora
Bugzilla. In case of problems, contact the owner of this repository.
Repository successfully enabled.
[root@localhost-live ~]# 
[root@localhost-live ~]# 
[root@localhost-live ~]# dnf install nvautoinstall -y
Copr repo for nvidia-auto-installer-for-fedora owned by t0xic0der                                                                                             2.6 kB/s | 2.5 kB     00:00    
Dependencies resolved.
==============================================================================================================================================================================================
 Package                         Architecture             Version                           Repository                                                                                   Size
==============================================================================================================================================================================================
Installing:
 nvautoinstall                   noarch                   0.4.1-2.fc38                      copr:copr.fedorainfracloud.org:t0xic0der:nvidia-auto-installer-for-fedora                    55 k

Transaction Summary
==============================================================================================================================================================================================
Install  1 Package

Total download size: 55 k
Installed size: 239 k
Downloading Packages:
nvautoinstall-0.4.1-2.fc38.noarch.rpm                                                                                                                          64 kB/s |  55 kB     00:00    
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Total                                                                                                                                                          63 kB/s |  55 kB     00:00     
Copr repo for nvidia-auto-installer-for-fedora owned by t0xic0der                                                                                             3.9 kB/s | 1.0 kB     00:00    
Importing GPG key 0x07D7D182:
 Userid     : "t0xic0der_nvidia-auto-installer-for-fedora (None) <t0xic0der#nvidia-auto-installer-for-fedora@copr.fedorahosted.org>"
 Fingerprint: 89F4 342E 2383 F5B2 6CF7 E529 C204 3BBB 07D7 D182
 From       : https://download.copr.fedorainfracloud.org/results/t0xic0der/nvidia-auto-installer-for-fedora/pubkey.gpg
Key imported successfully
Running transaction check
Transaction check succeeded.
Running transaction test
Transaction test succeeded.
Running transaction
  Preparing        :                                                                                                                                                                      1/1 
[... 1406 lines cut ...]
ttm                   114688  2 drm_ttm_helper,nouveau
drm_display_helper    237568  1 nouveau
cec                    86016  1 drm_display_helper
[root@localhost-live ~]# 
[root@localhost-live ~]# find /dev -group video
/dev/fb0
/dev/dri/card0
/dev/media0
/dev/video1
/dev/video0
[root@localhost-live ~]# 
[root@localhost-live ~]# cat /proc/cmdline
BOOT_IMAGE=(hd1,gpt2)/vmlinuz-6.2.0-0.rc2.20230105git41c03ba9beea.20.fc38.x86_64 root=UUID=f49ef0c5-b07a-4076-b79a-8c4211b480d5 ro rootflags=subvol=root rhgb quiet rd.driver.blacklist=nouveau modprobe.blacklist=nouveau nvidia-drm.modeset=1 initcall_blacklist=simpledrm_platform_driver_init
[root@localhost-live ~]# 
[root@localhost-live ~]# glxinfo | grep -i "vendor\|rendering"
direct rendering: Yes
server glx vendor string: SGI
client glx vendor string: Mesa Project and SGI
    Vendor: nouveau (0x10de)
OpenGL vendor string: nouveau
[root@localhost-live ~]# 
[root@localhost-live ~]# 
```

Thanks, for your support.

onuralpszr, 2023-01-08:
Can you check latest your latest installation log and post it here, 

Log location : /var/cache/akmods/nvidia/

Example log name :  525.60.11-1-for-6.0.16-300.fc37.x86_64.log

driver-version-"for"-kernel-version  <-- basically find your current most updated  kernel and driver version you installed. It should show what akmod did.

onuralpszr, 2023-01-08:
Issue is kernel/nvidia compatability and you are testing kernel so issue is not related to us. That is why I said I cannot guarantee your nvidia card nor this software in rawhide system

onuralpszr, 2023-01-08:
@t0xic0der close please

xxavi-hola-mon, 2023-01-08:
Hi @onuralpszr 

OK, thanks !

See you.

closed 2023-01-08 by xxavi-hola-mon