gridhead/nvidia-auto-installer-for-fedora-linux #270: Does not work on Fedora 39

mbledkowski, 2023-11-10:
When trying to install the package I get the following error:

```console
Error:
 Problem: conflicting requests
  - nothing provides python(abi) = 3.11 needed by nvautoinstall-0.4.1-2.fc38.noarch from copr:copr.fedorainfracloud.org:t0xic0der:nvidia-auto-installer-for-fedora
  - nothing provides python3.11dist(click) >= 7.1.2 needed by nvautoinstall-0.4.1-2.fc38.noarch from copr:copr.fedorainfracloud.org:t0xic0der:nvidia-auto-installer-for-fedora
  - nothing provides python3.11dist(distro) >= 1.5 needed by nvautoinstall-0.4.1-2.fc38.noarch from copr:copr.fedorainfracloud.org:t0xic0der:nvidia-auto-installer-for-fedora
```

onuralpszr, 2023-11-10:
It needs to be rebuild, it will be done

onuralpszr, 2023-11-13:
Rebuild order sent 

https://copr.fedorainfracloud.org/coprs/t0xic0der/nvidia-auto-installer-for-fedora/build/6627555/

closed 2023-11-19 by onuralpszr