gridhead/nvidia-auto-installer-for-fedora-linux #288: Getting "NVIDIA kernel module missing" despite following all instructions

sschuberth, 2024-01-20:
It seems that after some update of my Fedora 39 PC (KDE spin), I'm now at kernel 6.6.11-200.fc39.x86_64, GPU acceleration stopped working. So I went through all the installation steps, which succeeded, but still the start screen of KDE Plasma shows "NVIDIA kernel module not found. Falling back to Nouveau." even after several reboots and re-installation of `akmod-nvidia`. Any hints what I could check?

onuralpszr, 2024-01-20:
Is your secure boot on  ?

sschuberth, 2024-01-20:
Hmm, I was about to say "no" as this first shows "Secure boot disabled" (twice), but in the end it also shows "Secure boot is enabled"!?

```
dmesg | grep -iE 'secure|nouveau|nvidia'
[    0.000000] Command line: BOOT_IMAGE=(hd0,gpt4)/vmlinuz-6.6.11-200.fc39.x86_64 root=UUID=5493a814-c2ae-47b7-8a70-7b9b7b8cc117 ro rootflags=subvol=root rhgb quiet initcall_blacklist=simpledrm_platform_driver_init nvidia-drm.modeset=1 rd.driver.blacklist=nouveau modprobe.blacklist=nouveau
[    0.000000] secureboot: Secure boot disabled
[    0.041048] secureboot: Secure boot disabled
[    0.235871] Kernel command line: BOOT_IMAGE=(hd0,gpt4)/vmlinuz-6.6.11-200.fc39.x86_64 root=UUID=5493a814-c2ae-47b7-8a70-7b9b7b8cc117 ro rootflags=subvol=root rhgb quiet initcall_blacklist=simpledrm_platform_driver_init nvidia-drm.modeset=1 rd.driver.blacklist=nouveau modprobe.blacklist=nouveau
[   19.778159] Bluetooth: hci0: Secure boot is enabled
```

sschuberth, 2024-01-20:
> Is your secure boot on ?

I've double-checked in UEFI settings, and it's disabled.

sschuberth, 2024-01-22:
The problem magically disappeared with today's kernel update to version `6.6.12-200.fc39.x86_64`.

closed 2024-01-22 by sschuberth

sschuberth, 2024-04-27:
This happening again after the upgarde from Fediopra 39 to 40. What helped was:

```
sudo dnf remove kmod-nvidia-6.8.7-300.fc40.x86_64.x86_64
sudo dnf reinstall akmod-nvidia
```

FelipeRuizGarcia, 2024-04-27:
Hello @sschuberth , glad to see the issue goes away . So, I saw the same issue upgrading the kernel. I fixed just reinstalling the pieces around. This is the script i use to do it.

https://github.com/FelipeRuizGarcia/super-machine-setup/blob/test/utils/fedora/os-fedora-nvidia.sh