gridhead/nvidia-auto-installer-for-fedora-linux #307: does this installer work with NVIDIA GeForce GT 755M  ?

goldenratio, 2024-05-08:
I have installed all modes -  driver, nvrepo, plcuda, ffmpeg, vulkan, vidacc, primec

It seems GPU is not detected? 

Nvidia X Server settings doesn't show the GPU,

[image]

nvtop and nvidia-semi doesn't detect GPU,
[image]

Any help is greatly appreciated.

neofetch log,
```
             .',;::::;,'.                femi@b6de-6e3d-be7a-af67-9800-842c-07d 
         .';:cccccccccccc:;,.            -------------------------------------- 
      .;cccccccccccccccccccccc;.         OS: Fedora Linux 40 (Workstation Editi 
    .:cccccccccccccccccccccccccc:.       Host: 20217 Lenovo IdeaPad Y510P 
  .;ccccccccccccc;.:dddl:.;ccccccc;.     Kernel: 6.8.8-300.fc40.x86_64 
 .:ccccccccccccc;OWMKOOXMWd;ccccccc:.    Uptime: 26 mins 
.:ccccccccccccc;KMMc;cc;xMMc:ccccccc:.   Packages: 2488 (rpm), 34 (flatpak) 
,cccccccccccccc;MMM.;cc;;WW::cccccccc,   Shell: bash 5.2.26 
:cccccccccccccc;MMM.;cccccccccccccccc:   Resolution: 2560x1440 
:ccccccc;oxOOOo;MMM0OOk.;cccccccccccc:   DE: GNOME 46.1 
cccccc:0MMKxdd:;MMMkddc.;cccccccccccc;   WM: Mutter 
ccccc:XM0';cccc;MMM.;cccccccccccccccc'   WM Theme: Adwaita 
ccccc;MMo;ccccc;MMW.;ccccccccccccccc;    Theme: Adwaita [GTK2/3] 
ccccc;0MNc.ccc.xMMd:ccccccccccccccc;     Icons: Adwaita [GTK2/3] 
cccccc;dNMWXXXWM0::cccccccccccccc:,      Terminal: gnome-terminal 
cccccccc;.:odl:.;cccccccccccccc:,.       CPU: Intel i7-4700MQ (8) @ 3.400GHz 
:cccccccccccccccccccccccccccc:'.         GPU: NVIDIA GeForce GT 755M 
.:cccccccccccccccccccccc:;,..            GPU: NVIDIA GeForce GT 755M 
  '::cccccccccccccc::;,.                 Memory: 4394MiB / 15931MiB 

```

FelipeRuizGarcia, 2024-05-08:
Hello @goldenratio , 

Questions, 
1. is uefi disable on the machine ? if not, can you disable ?
2. have been worked in that before in that machine ? 

I have been using a Lenovo with fedora 40 as well, saw the same issue as you upgrading the kernel.
The issues reinstalling some packages. I create this script to do it: 

https://github.com/FelipeRuizGarcia/super-machine-setup/blob/test/utils/fedora/os-fedora-nvidia.sh

that work for me. Hopefully can help u.

goldenratio, 2024-05-09:
- is uefi disable on the machine ? if not, can you disable ?
ya, UEFI is disable and also secure boot is disabled

- have been worked in that before in that machine ?
it has never worked

reinstalling didn't work

FelipeRuizGarcia, 2024-08-15:
@goldenratio , take a look to the kernel modules and specifc versions for your graphics card.

https://rpmfusion.org/Howto/NVIDIA

FelipeRuizGarcia, 2024-08-16:
@goldenratio , take a look to the kernel modules and specifc versions for your graphics card.

https://rpmfusion.org/Howto/NVIDIA 

Lets see if the nvidia kernel modules are loaded .
share with us the `cat /default/grub`

there should be an `nvidia-drm.modeset=1` option to be able to load

also, `sudo lsmod | grep nvidia`

I am ready to help you with this.