gridhead/nvidia-auto-installer-for-fedora-linux #330: Create builds for Fedora 41

sschuberth, 2024-10-30:
Looking at https://copr.fedorainfracloud.org/coprs/t0xic0der/nvidia-auto-installer-for-fedora/build/6627555/, the `fedora-41-x86_64` entry says "RPM build has not started yet", and following the link to [here](https://download.copr.fedorainfracloud.org/results/t0xic0der/nvidia-auto-installer-for-fedora/fedora-41-x86_64/06627555-nvautoinstall/) shows `*-fc40.*.rpm` file names, although it should probably be "41" instead of "40".

gridhead, 2024-10-31:
Roger. Getting to it.

gridhead, 2024-10-31:
@sschuberth the new releases are now available here https://copr.fedorainfracloud.org/coprs/t0xic0der/nvidia-auto-installer-for-fedora/build/8192842/.

Ref. https://github.com/gridhead/nvidia-auto-installer-for-fedora-linux/releases/tag/v0.4.2

closed 2024-10-31 by gridhead