gridhead/nvidia-auto-installer-for-fedora-linux #334: Code is messy and largely broken

JosephM101, 2024-12-10:
I'm writing this not to harshly criticize your work, but to inform you that your current tree on the master branch contains numerous syntax and formatting issues that result in certain functions of the script not looking right or functioning properly.

Additionally, cryptic function names (such as `rpin` and `rpck`) make understanding the code much more difficult. It also rings a bit like a decompilation.

I am interested in and considering creating a pull request where I fix these issues. Changing the function names to make more sense, and properly formatting some of the broken code to alleviate issues.

An example of such an issue was with the `rpmadd` command. The `$(rpm -E %fedora)` substitution was formatted improperly resulting in an HTTP 404 error and subsequent script failure. I fixed the formatting, installed the modified code on my machine and it ran perfectly. This would need to be done to other parts of the code to fix these kinds of issues and improve readability and consistency.

@gridhead If you'd like, I can take some extra time (should I come across any) to fix these issues for you and submit a pull request.

JosephM101, 2024-12-10:
For context, this is what I fixed:

#### nvautoinstall/operations/handle_rpmfusion_repositories.py
```
def main(self):
        os.system("dnf install -y fedora-workstation-repositories")
        os.system(
            "dnf install -y"
            " https://mirrors.rpmfusion.org/free/fedora/rpmfusion-free-release-$(rpm -E %fedora).noarch.rpm"
        )
        os.system(
            "dnf install -y"
            " https://mirrors.rpmfusion.org/nonfree/fedora/rpmfusion-nonfree-release-$(rpm -E %fedora).noarch.rpm"
        )

        ...
```

gridhead, 2024-12-11:
All good. @onuralpszr and I plan on rewriting the codebase with the native compatibility of libdnf5, as marked in #320. I am going to go ahead and close this one so that we can continue our discussions over there on this. Thanks for letting us know about the same.

closed 2024-12-11 by gridhead